LinuxCNC/linuxcnc #2453: Gmoccapy MSG: Must be in MDI mode.....

zz912, 2023-04-27:
I use RIP LCNC Branche 2.9

Simulate the problem:
1)    edit /home/user/linuxcnc/linuxcnc-2.9/configs/sim/gmoccapy/gmoccapy.ini
    add:

```
[HALUI]
MDI_COMMAND = M61 Q5
MDI_COMMAND = M61 Q2
```

2) Run LCNC
[image]

3) This error is random, so it may not appear the first time.
Constantly switch to MDI mode and enable/disable HAL pins halui.mdi-command-00 and halui.mdi-command-01

MDI commands run fine, but sometimes this message pops up.

I have more sophisticated commands on the real machine. I was in MDI-mode the whole time when I tuned them and never had this problem. However, for normal use of my commands, I want to be in JOG mode and that's where the message appears.

gmoccapy, 2023-04-28:
As there are three modes, Manual and MDI and AUTO the behavior of gmoccapy is exactly as it should!

No MDI commands in MANUAL mode.
To execute commands from MANUAL mode, you could combine MDI commands, first command switch to MDI then execute the MDI command you want and switch back to manual mode. You may use the corresponding Halui commands to change Modes.

Doing this, may result in a short flicker of the screen, as gmoccapy will change the screen design due to the mode switch.

Norbert

zz912, 2023-04-28:
Hello Norbert,

1) Thanks for explaining the problem. Now I know how to solve it.

2) I think Gmoccapy behavior is not ideal. What bothers me the most is that he acts randomly. Paradoxically, more often it allows you to execute an MDI command without this message popping up, and every time the MDI commands are executed in AUTO. It is not absolutely necessary, but it would be nice if the message appeared regularly and the execution of MDI commands was really prohibited.

3) Would it be possible to add a "HAL pin" to Gmoccapy that would pause the redraw of the Gmoccapy screen? I could make sure that the windows wouldn't flash short flicker of the screen.

Or would it be possible to add a "HAL pin" to Gmoccapy that would allow make MDI execution in every modes?

Or is there any other solution to prevent short flicker of the screen?

Zdeněk

zz912, 2023-04-28:
I see fight between Gmoccapy and Halui.

Gmoccapy wants "No MDI commands in MANUAL mode."

But Halui sets MDI mode for MDI_COMMAND event.
Look here:
https://github.com/LinuxCNC/linuxcnc/blob/46557dc6245fd685a6d44d37d21f8b1458ff28d0/src/emc/usr_intf/halui.cc#L1079
(I hope, that I understood halui code correct)

First we should define the correct LCNC behavior when changing the halui.mdi-command-XX pin

hansu, 2023-04-29:
[Norbert]
>To execute commands from MANUAL mode, you could combine MDI commands, first command switch to MDI then execute the MDI command you want and switch back to manual mode. You may use the corresponding Halui commands to change Modes.

It seems that _halui_ is doing that already.

[Zdeněk]
I tested that several times and I don't get that message.
When I run the MDI command in manual mode, LinuxCNC switches to MDI mode, runs the command and then switch back to Manual mode. Indeed some flickering especially if you have enabled the on-screen keyboard.

zz912, 2023-04-29:
[HansU]
Can I ask for modify test?
```
[HALUI]
MDI_COMMAND = M61 Q5 G4 P0.5
MDI_COMMAND = M61 Q2 G4 P0.5
```
Now I tried the test without the G4 on a third PC and the problem did not appear like you. With the G4, it will show, but only for the twentieth time.

zz912, 2023-04-30:
I tested it with Axis and there were not any problems.
[image]

hansu, 2023-04-30:
I still don't get that message. Does is appear when you set the execute command pin or on startup?

zz912, 2023-04-30:
I will make a video of the screen tonight, how to simulate the error.

zz912, 2023-04-30:
Here is video:
https://user-images.githubusercontent.com/96618597/235367645-18399092-23a7-4bb3-8516-3df2d3b17171.mp4

The bug behaves very randomly. On this video, the message occurs more often than in other cases. Maybe the video recorder supports this bug. Even so, you can see the randomness in the video. I usually have to click through Set and Clr far more often to get it to appear at least once. Furthermore, I found that increasing the probability of a bug occurring can be ensured by increasing the number of G and M codes in MDI_COMMAND.

hansu, 2023-04-30:
The video file seems to be damaged. Please upload again.

zz912, 2023-04-30:
The video is OK. It works in some player and in some player not. Can I ask you for VLC player use?

I test it in 2 PC and it works. In my mobile it works not.

hansu, 2023-04-30:
Ok it works if I download it. Just didn't work in the browser. Furthermore Github is now capable of embedding videos.

zz912, 2023-05-03:
Hi Hans,

did you manage to simulate this bug?

hansu, 2023-05-04:
No I still couldn't reproduce it. But maybe that's because I am running it on a VM. I'll have another try on a real machine...

zz912, 2023-05-04:
That's weird. I am able to simulate it on 2 PCs and 1 VM. Try reducing the CPU on the VM. I have a feeling that the error is more frequent when the CPU is more heavily loaded. Would it help if I gave you access to a PC?

rene-dev, 2023-05-04:
the three modes, Manual, MDI and AUTO are not a feature of gmocappy, but of linuxcnc task. I believe you do not see the behavior in axis, as it does not switch the ui when task changes modes.

zz912, 2023-05-05:
[rene-dev]
> the three modes, Manual, MDI and AUTO are not a feature of gmocappy, but of linuxcnc task. I believe you do not see the behavior in axis, as it does not switch the ui when task changes modes.

You're right. https://github.com/LinuxCNC/linuxcnc/issues/2453#issuecomment-1529031686

[Norbert]
> Try reducing the cycle time (INI setting), Default is 100 , try to set the value to 150 or 200
> 
> I could reproduce the error, but only on one PC (poor CPU Power) setting the cycle time to 150 solved the problem on my PC.
> 
> Norbert

I tried 300ms and it did not help. :-(

I would like to show you one more thing about this bug. Sometimes Gmoccapy gets stuck in MDI mode. Watch this video. At time 00:00 Gmoccapy is in MANUAL mode and then after executing MDI_COMMAND it remains in MDI mode.

https://user-images.githubusercontent.com/96618597/236371597-ba34711e-0e33-4a97-a3a4-c2f8a7115e6d.mp4

In this video CYCLE_TIME =  300

gmoccapy, 2023-05-05:
Gmoccapy do change the screen Design according to the MODE selection button or due to signals from LinuxCNC and so far it works as it should.

The problem seems to be related to the MDI commands. If you try with an MDI command with any movement i.e. G91 G0 Z0.001 you will not be able to reproduce the error, as LinuxCNC will emit the signal of the actual Mode but if you use a command without a Gcode Brake the signal will not be emmited.

I do not know a solution at this moment. 
I will finish my house building in about 3 to 6 month so agter that I will have more time to look at that kind of behavior.

Norbert

zz912, 2023-05-05:
I tried:
```
[HALUI]
MDI_COMMAND = M61 Q5  G91 G0 Z-5.001
MDI_COMMAND = M61 Q2  G91 G0 Z-10.001
```
It did not help. I don't need this bug fixed immediately, but I could ask you to label this bug 2.9-must-fix, or add it to some list of things that must be fixed.

zz912, 2023-05-07:
I tried to do the same with TAG 2.8.4

1) edit /home/zdenek/linuxcnc/linuxcnc-2.8.4/configs/sim/gmoccapy/gmoccapy.ini
add:
```
[HALUI]
MDI_COMMAND = M61 Q5  G91 G0 Z-5.001
MDI_COMMAND = M61 Q2  G91 G0 Z-10.001
```

2) edit /home/zdenek/linuxcnc/linuxcnc-2.8.4/configs/sim/gmoccapy/gmoccapy_postgui.hal
add:
```
net pokus-00  halui.mist.is-on halui.mdi-command-00
net pokus-01  halui.flood.is-on halui.mdi-command-01
```
In the LCNC 2.8 version, the SET and CLR buttons are not in the halshow, so I used the MIST and FLOOD buttons.

3) Run LCNC

The video shows that the error also appears in version 2.8.4.
https://user-images.githubusercontent.com/96618597/236683209-0ad5658a-c7f5-49a8-ae24-304b169c125c.mp4

zz912, 2023-05-07:
I did the test from the previous post on other versions:
linuxcnc-2.7.15 - works good without message "Must be in MDI mode to issue MDI command"
linuxcnc-2.8.0 - bug
linuxcnc-2.8.2 - bug
linuxcnc-2.8.3 - bug
linuxcnc-2.8.4 - bug
linuxcnc-2.9 - bug

Finding this error is challenging. Appears randomly.

Here is the terminal listing when I press the button and execute MDI_COMMAND.

LCNC 2.7.15:
```
Emit interp-run
3 2
Emit interp-run
```

LCNC 2.8.0:
```
MANUAL Mode
IDLE
hal status motion mode changed
```
LCNC 2.8.0 bug:
```
3 2
('MDI Mode', False)
RUN
hal status motion mode changed
Must be in MDI mode to issue MDI command
MANUAL Mode
IDLE
hal status motion mode changed
```

[HansU]
Have you tried lowering the CPU frequency in the VM to simulate this bug? Norbert wrote that he managed to simulate it on one PC (poor CPU Power), maybe it would help you to see this bug. If you can't simulate this bug, I could send you my VM, but I won't have access to it until Tuesday.

zz912, 2023-05-09:
I asked Fupe to try to look into this problem of mine.

At first I was disappointed because his feedback was that he was unable to simulate the problem on either a physical PC or a Virtual Machine.

Finally we found out that he is not using Oracle VM, but using another VM. So based on his help, we know that we need to use Oracle VM to simulate this bug.

Next, I verified that it is really necessary to reduce the CPU performance. At 100% CPU the error did not appear, at 30% the error appears stably.

[image]

hansu, 2023-05-09:
No, I still don't get that message even when limiting the CPU to 30 % :/

zz912, 2023-05-09:
Thank you for info.

zz912, 2023-05-10:
Hello,

I am very unhappy that you are not able to simulate my bug. I made another attempt today.

I made a new Virtual Box in Oracle VM. I installed it with linuxcnc-2.8.2-buster.iso file.

In order to install this iso file, it is necessary to set at least 2 processor cores, otherwise the installation will fail.

After installing I updated:
```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-update
```

I started linuxcnc and opened the Gmoccapy sim.

I closed linuxcnc

edit /home/zdenek/linuxcnc/configs/sim.gmoccapy/gmoccapy.ini
add:
```
[HALUI]
MDI_COMMAND = M61 Q5 G91 G0 Z-5.001
MDI_COMMAND = M61 Q2 G91 G0 Z-10.001
```
edit /home/zdenek/linuxcnc/configs/sim.gmoccapy/gmoccapy_postgui.hal
add:
```
net attempt-00 halui.mist.is-on halui.mdi-command-00
net attempt-01 halui.flood.is-on halui.mdi-command-01
```

Now when I tried to simulate the bug, the bug did not appear.

Then I reduced the CPU performance and the bug appeared.

[hansu]
Can I ask for another test with lower CPU performance?

Here you can download the VM file for Oracle VM VirtualBox:
https://ulozto.cz/tamhle/cVRcwtDslND0#!ZGSvLmR2Awx5LmNjMJWwBGWyAQx3LGORGzkHFaqPEUEPEGZkLD==
login: zdenek
password: dedadeda

I did the installation twice. Once in Czech and once in English. There was no difference.

zz912, 2023-05-10:
Is there an answer to your question?
[image]

If you think of anything else I could try, let me know.

If you still can't simulate it, I plan to try it on an old physical PC with bad latency and mail it to you.

hansu, 2023-05-10:
No nothing :(
I tried down to 2 cores, 20% cpu limit where the OS on the VM is almost not responding, but no error message.
I have an old PC I can install LinuxCNC on and I can also try ob my laptop which only have a Core i5.
But not sure when I'll have time for this.

zz912, 2023-05-11:
Can I ask you if you could try changing the INI like this:
```
[HALUI]
MDI_COMMAND = M61 Q5  G91 G0 Z-5.001  G4 P5
MDI_COMMAND = M61 Q2  G91 G0 Z-10.001  G4 P5
```
?

Stay in VM with low CPU.

I found that the more commands in MDI_COMMANDS, the more occurrences of that message.

Here it is even interesting that sometimes it ignores the G4 command and does not give any message.

hansu, 2023-05-14:
If it's mainly the message that bothers you - I can write you a fix which is suppressing this message.

zz912, 2023-05-14:
I would like to ask you if you could look at this bug first:
https://github.com/LinuxCNC/linuxcnc/issues/2489
Maybe it's related.

I don't like to hide errors, I will check if it has functional errors as well.

I looked here:
https://github.com/LinuxCNC/linuxcnc/blob/e2113f0c5b949f52c4e6ca35caa9b0395fa8a655/src/emc/usr_intf/gmoccapy/gmoccapy.py#L2769
I think that the creation of Gmoccapy did not take into account the fact that halui changes its mode by itself.

c-morley, 2023-05-14:
Can you find in Gmoccapy's code where the message originates from?
How does it decide that an MDI command was called when in manual mode?
It's odd that gmoccapy is detecting an MDI move originated from something else - I'm not sure it can.
I think its is detecting the mode change using some object that is slightly out of sync with the motion controller (race condition)

hansu, 2023-05-14:
>I don't like to hide errors, I will check if it has functional errors as well.

I wanted to say "I can write you a **temporary** fix which is suppressing this message **until this bug gets fixed**.

>Can you find in Gmoccapy's code where the message originates from?

The message is not originated from gmoccapy - it is coming from here: https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/emc/task/emctaskmain.cc#L2215

hansu, 2023-05-14:
>I looked here:
>
>[linuxcnc/src/emc/usr_intf/gmoccapy/gmoccapy.py](https://github.com/LinuxCNC/linuxcnc/blob/e2113f0c5b949f52c4e6ca35caa9b0395fa8a655/src/emc/usr_intf/gmoccapy/gmoccapy.py#L2769)
>
>Line 2769 in [e2113f0](https://github.com/LinuxCNC/linuxcnc/commit/e2113f0c5b949f52c4e6ca35caa9b0395fa8a655)
>`# if MDI button is not sensitive, we are not ready for MDI commands `
>
>I think that the creation of Gmoccapy did not take into account the fact that halui changes its mode by itself.

`on_hal_status_mode_mdi(self, widget)` is called when the mode is changed, e.g. by halui.
Sensitive here means that they are not grayed out which would be the case when the machine is turned off. 
When MDI mode is active the button has the state "checked".

zz912, 2023-05-14:
Thanks Chriss for joining.

> Can you find in Gmoccapy's code where the message originates from?

I think (90 percent) that Gmoccapy is causing this message, but this message is not in Gmoccapy, but here:
https://github.com/LinuxCNC/linuxcnc/blob/494b5316e2f4c5f865985dbeec2d04b8ec4d5afd/src/emc/task/emctaskmain.cc#L2215

> How does it decide that an MDI command was called when in manual mode?

This is probably the source of the problem.
HALUI can call MDI_COMMAND at any time, in any mode.
http://linuxcnc.org/docs/2.9/html/config/ini-config.html#sub:ini:sec:halui

HALUI set MDI mode here:
https://github.com/LinuxCNC/linuxcnc/blob/46557dc6245fd685a6d44d37d21f8b1458ff28d0/src/emc/usr_intf/halui.cc#L1079

After execute commands HALUI set old mode here:
https://github.com/LinuxCNC/linuxcnc/blob/46557dc6245fd685a6d44d37d21f8b1458ff28d0/src/emc/usr_intf/halui.cc#L2009

I think Gmoccapy doesn't assume HALUI functionality and fights it.

[HansU]
> I wanted to say "I can write you a temporary fix which is suppressing this message until this bug gets fixed.

I am doing a retrofit for a friend who is not pressed for time. You don't have to do some temporary solution. Thank you for offer.

zz912, 2023-05-14:
[HansU]
> Sensitive here means that they are not grayed out which would be the case when the machine is turned off.

Not only the case when the machine is turned off. Look:
https://github.com/LinuxCNC/linuxcnc/assets/96618597/b0a4c63e-fb1f-4500-a0d6-75ed91832b21

c-morley, 2023-05-14:
It's sounding like the error is not with gmoccapy but maybe with halui.
I can't see where halui confirms  the mode switch happened before issuing the MDI command.

I wonder if on most systems the mode switch is fast enough the there is no problem.
on  a low powered machine using a high resource gui (gmoccapy) it shows the problem.

Could test by adding a delay to halui right after the mode switch  and see if that fixes it.

hansu, 2023-05-14:
@zz912 Suggestion to record the screen: https://github.com/phw/peek - you can install it simply via the package repositories.

zz912, 2023-05-15:
At the moment esleep really works. The esleep function parameter is in seconds and I see that I have to wait 1s + 1s for everything to happen.
[image]
Unfortunately, this did not fix the bug.

zz912, 2023-05-18:
Hello every body,

I found the bug in source code !!!!!!!!!!!!! I am so happy!!!!!!!!

Chris was very close to solving the problem. Needed to add sleep before returning mode.
[image]
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/usr_intf/halui.cc#L2133

When I first tried esleep, I tried it with a value of 0.1s = 100ms. The probability of occurrence of the message has decreased, but the message has not disappeared. By gradually increasing the value, I reached the value of 0.4s = 400ms. At this value, it passed the 10 minute click test without a bug.

I would like to ask you to suggest a better solution than the "sleep process". I would like something "if MDI_COMMAND == finished { continue }"

This condition may not work correctly:
`if (emcStatus->status == 1) { //which seems to have finished`

zz912, 2023-05-21:
Hello,

Today I wanted to continue looking for the bug. I started by downloading the current 2.9 branch. I found that the bug is gone.

I looked up what fixed the bug.

I found that this commit https://github.com/LinuxCNC/linuxcnc/commit/bdb442a3dfd451ab482b9950fb6b7e507dbdb7a6 solved the problem.

I would like to thank SebKuzminsky for fixing the cause of the bug and others for their help.

hansu, 2023-05-21:
Great!

zz912, 2023-05-21:
I apologize for the confusion. This bug was not repaired. Bug behavior is random and malicious. I made over 20 attempts and she didn't show up. So I documented it. Then I checked whether this commit affected other bugs as well. I figured out that this bug just didn't appear.

zz912, 2023-05-22:
So today I devoted myself to finding this bug again.

I modified this part of the code:
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/usr_intf/halui.cc#L2133

```
    if (halui_sent_mdi) { // we have an ongoing MDI command
		int array[10];
		for (int i = 0; i < 10; i++){
			array[i] = emcStatus->status;
			esleep(0.001); //sleep for a while
			}
		        
        if (emcStatus->status == 1) { //which seems to have finished
			halui_sent_mdi = 0;
			switch (halui_old_mode) {
			case EMC_TASK_MODE_MANUAL: 
				fprintf(stderr,"HAF HAF MANUAL MODE will be set\n");
				sendManual();
				fprintf(stderr,"HAF HAF MANUAL MODE was set\n");
				break;
			case EMC_TASK_MODE_MDI: break;
			case EMC_TASK_MODE_AUTO: sendAuto();break;
			default: sendManual();break;
			}
		}
			    
	    fprintf(stderr,"\nHAF HAF emcStatus->status:\n");
		for (int i = 0; i < 10; i++){        
			fprintf(stderr,"HAF HAF %d\n",array[i]);
			}
		fprintf(stderr,"\n");
    }
```
Result without bug:
```
3 2

HAF HAF emcStatus->status:
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2

HAF HAF emcStatus->status:
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2

HAF HAF emcStatus->status:
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2

HAF HAF emcStatus->status:
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2

HAF HAF MANUAL MODE will be set
HAF HAF MANUAL MODE was set

HAF HAF emcStatus->status:
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
```

Result with bug:
```
3 2

HAF HAF emcStatus->status:
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2

HAF HAF emcStatus->status:
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2

HAF HAF emcStatus->status:
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2
HAF HAF 2

HAF HAF MANUAL MODE will be set
Must be in MDI mode to issue MDI command
HAF HAF MANUAL MODE was set

HAF HAF emcStatus->status:
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1
HAF HAF 1

```

1) So far it looks like the bug occurs after executing this command:  https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/usr_intf/halui.cc#L2135
2) The emcStatus->status value is stable.
3) The emcStatus->status value will change to 1 very soon.

I still can't think of a way to solve it. I don't like solutions with a large esleep value.

petterreinholdtsen, 2023-05-25:
[zz912]
> I still can't think of a way to solve it. I don't like solutions with
> a large esleep value.

It sound like there is a concurrency/race condition where two tasks need
to be done in a specific sequence which is currently not guaranteed by
the code.  Adding the sleep to the task that need to be done last will
hide the problem, as it increase the chance of the first task being done
before the last task is running.

Do you know whihc tasks / code paths are involved?  Can some locking
mechanism (like mutual exlusion / mutex) be added to ensure a
predictable ordering?

-- 
Happy hacking
Petter Reinholdtsen

zz912, 2023-05-25:
I added comments here:
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/usr_intf/halui.cc#L2135

```
   if (halui_sent_mdi) { // we have an ongoing MDI command
        if (emcStatus->status == 1) { //which seems to have finished
			halui_sent_mdi = 0;
			switch (halui_old_mode) {
			case EMC_TASK_MODE_MANUAL: 
				fprintf(stderr,"***********************************\n");
				fprintf(stderr,"HAF HAF MANUAL MODE will be set\n");
				fprintf(stderr,"***********************************\n");
				sendManual();
				fprintf(stderr,"***********************************\n");
				fprintf(stderr,"HAF HAF MANUAL MODE was set\n");
				fprintf(stderr,"***********************************\n");
				break;
			case EMC_TASK_MODE_MDI: break;
			case EMC_TASK_MODE_AUTO: sendAuto();break;
			default: sendManual();break;
			}
		}
    }
```
I found the "emcStatus->status" definition here:
https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/task/emctaskmain.cc#L3551
https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/task/emctaskmain.cc#L3561
https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/task/emctaskmain.cc#L3564

RCS_DONE: 1
RCS_EXEC: 2
RCS_ERROR: 3

So, I added comments here:
https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/task/emctaskmain.cc#L3572

```
	emcStatusBuffer->write(emcStatus);
	

	printf("!taskPlanError: %i\n", !taskPlanError);
	printf("!taskExecuteError: %i\n", !taskExecuteError);
	printf("emcStatus->task.execState == EMC_TASK_EXEC_DONE: %i\n", emcStatus->task.execState == EMC_TASK_EXEC_DONE);
	printf("emcStatus->motion.status == RCS_DONE: %i\n", emcStatus->motion.status == RCS_DONE);	
	printf("emcStatus->io.status == RCS_DONE: %i\n", emcStatus->io.status == RCS_DONE);
	printf("mdi_execute_queue.len() == 0: %i\n", mdi_execute_queue.len() == 0);
	printf("interp_list.len() == 0: %i\n", interp_list.len() == 0);
	printf("emcTaskCommand == 0: %i\n", emcTaskCommand == 0);
	printf("emcStatus->task.interpState == EMC_TASK_INTERP_IDLE: %i\n", emcStatus->task.interpState == EMC_TASK_INTERP_IDLE);		
	printf("HAF HAF emcStatus->status: %i\n\n\n", emcStatus->status);
```

Result without bug:
1) emcStatus->status: 1
2) set manual mode
3) emcStatus->status: 2 (only 2 times)
4) emcStatus->status: 1 (many times)
5) set manual mode finish

```
!taskPlanError: 1
!taskExecuteError: 1
emcStatus->task.execState == EMC_TASK_EXEC_DONE: 1
emcStatus->motion.status == RCS_DONE: 1
emcStatus->io.status == RCS_DONE: 1
mdi_execute_queue.len() == 0: 1
interp_list.len() == 0: 1
emcTaskCommand == 0: 1
emcStatus->task.interpState == EMC_TASK_INTERP_IDLE: 1
HAF HAF emcStatus->status: 1
[... 1015 lines cut ...]
emcStatus->task.execState == EMC_TASK_EXEC_DONE: 1
emcStatus->motion.status == RCS_DONE: 1
emcStatus->io.status == RCS_DONE: 1
mdi_execute_queue.len() == 0: 1
interp_list.len() == 0: 1
emcTaskCommand == 0: 1
emcStatus->task.interpState == EMC_TASK_INTERP_IDLE: 1
HAF HAF emcStatus->status: 3

!taskPlanError: 0
!taskExecuteError: 1
emcStatus->task.execState == EMC_TASK_EXEC_DONE: 1
emcStatus->motion.status == RCS_DONE: 1
emcStatus->io.status == RCS_DONE: 1
mdi_execute_queue.len() == 0: 1
interp_list.len() == 0: 1
emcTaskCommand == 0: 1
emcStatus->task.interpState == EMC_TASK_INTERP_IDLE: 1
HAF HAF emcStatus->status: 3

***********************************
HAF HAF MANUAL MODE was set
***********************************

```

zz912, 2023-05-25:
I continued to search for the bug. I edited this part:
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/usr_intf/halui.cc#L450
```
static int emcCommandSend(RCS_CMD_MSG & cmd)
{
    // write command
    if (emcCommandBuffer->write(&cmd)) {
        rtapi_print("halui: %s: error writing to Task\n", __func__);
        return -1;
    }
    emcCommandSerialNumber = cmd.serial_number;

    // wait for receive
    double end;
    for (end = 0.0; end < receiveTimeout; end += EMC_COMMAND_DELAY) {
	

	updateStatus();
	fprintf(stderr,"HAF HAF 1;\n");	
	
	int serial_diff = emcStatus->echo_serial_number - emcCommandSerialNumber;

	if (serial_diff >= 0) {
	    return 0;
	}
	fprintf(stderr,"HAF HAF 2;\n");
	esleep(EMC_COMMAND_DELAY);
	fprintf(stderr,"HAF HAF 3;\n");
    }

    rtapi_print("halui: %s: no echo from Task after %.3f seconds\n", __func__, receiveTimeout);
    return -1;
}
```

Result:
```
HAF HAF 1;
HAF HAF 2;
Must be in MDI mode to issue MDI command
HAF HAF 3;
HAF HAF 1;
```
Possible interpretation of the problem:
1) request to change the mode to MANUAL MODE
2) esleep will give the option to run emctaskmain.cc

zz912, 2023-05-27:
Hello everybody,

1) Please ignore recent posts.
I wanted to find a bug so that if a bug appears between two of my "HAF HAF" comments, that's where something is wrong. This method of finding bugs has worked for me in the past. Unfortunately, it is absolutely useless for finding race bugs. I had to start looking for the bug all over again.

2) To explain the bug, we'll start from the beginning. We will use a clean install of LCNC 2.8 and above.

3) edit /home/user/linuxcnc/linuxcnc-2.9/configs/sim/gmoccapy/gmoccapy.ini
add:
[HALUI]
MDI_COMMAND = M61 Q5
MDI_COMMAND = M61 Q2

4) It is very important that MDI_COMMAND contains the automatic tool command. Otherwise, the bug will not happen.

5) Now I will describe the race without bug:

We will change mode to MANUAL

HALUI: change mode to MDI
GMOCCAPY: -

HALUI: start MDI_COMMAND
GMOCCAPY: -

HALUI: execute command M61 Q5
GMOCCAPY: run on_hal_status_tool_in_spindle_changed https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/usr_intf/gmoccapy/gmoccapy.py#L2643-L2647

HALUI: finish MDI_COMMAND
GMOCCAPY: -

HALUI: change mode to old_mode (MANUAL)
GMOCCAPY: -

HALUI: -
GMOCCAPY: run _update_toolinfo
https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/usr_intf/gmoccapy/gmoccapy.py#L3497

HALUI: -
GMOCCAPY: run G43 + change MDI mode
https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/usr_intf/gmoccapy/gmoccapy.py#L3540-L3544

6) Now I will describe the race with bug:

We will change mode to MANUAL

HALUI: change mode to MDI
GMOCCAPY: -

HALUI: start MDI_COMMAND
GMOCCAPY: -

HALUI: execute command M61 Q5
GMOCCAPY: run on_hal_status_tool_in_spindle_changed

HALUI: -
GMOCCAPY: run _update_toolinfo

Now BOOM emctaskmain get command from HALUI for finish MDI_COMMAND + change mode to old_mode (MANUAL) and at the same time emctaskmain get command from GMOCCAPY run G43 + change MDI mode
Result this BOOM is message: Must be in MDI mode to issue MDI command

7) Theoretical solution
Remove this lines:
https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/usr_intf/gmoccapy/gmoccapy.py#L3540-L3544
As soon as I deleted these lines the bug did not appear. This solution is only suitable for testing my theory.

8) A suggested solution
"def on_hal_status_tool_in_spindle_changed(self, object, new_tool_no): " should be activated only when the main task is IDLE

9) Next problem
These lines leave MDI mode.
https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/usr_intf/gmoccapy/gmoccapy.py#L3540-L3544
The edit should look something like this:
```
       if "G43" in self.active_gcodes and self.stat.task_mode != linuxcnc.MODE_AUTO:
            old_mode = self.stat.task_mode
            self.command.mode(linuxcnc.MODE_MDI)
            self.command.wait_complete()
            self.command.mdi("G43")
            self.command.wait_complete()
            self.command.mode(old_mode)
            self.command.wait_complete()
```

I hope I managed to explain the bug.

zz912, 2023-05-28:
I looked in the history and the lines executing the G43 command in _update_toolinfo have been in Gmoccapy since the beginning.

Commits on Feb 28, 2014
https://github.com/LinuxCNC/linuxcnc/commit/579797703e6da6f026ed03896cad78be191b3dec#diff-f312cf11426f38ce14f58116be820e007813335ec2fc89eab9bcc7a6913808a9
https://github.com/LinuxCNC/linuxcnc/blob/579797703e6da6f026ed03896cad78be191b3dec/src/emc/usr_intf/gmoccapy/gmoccapy.py#L1950-L1955

And this control is even older than Gmoccapy.
https://github.com/LinuxCNC/linuxcnc/blob/eea3b99b90a05bac68bef549042c208905d35720/src/emc/task/emctaskmain.cc#L2016-L2019

It's strange that the bug didn't show up earlier.

I am thinking about how to make a fix for this bug. I would like to ask for an explanation:
What are the lines executing the G43 command in _update_toolinfo for?

I can think of several solutions to this bug, but I don't want to create more bugs with my fix. Anyway, I don't think it's a good idea to execute any commands in the "toolinfo" function.

zz912, 2023-05-31:
I tried to modify Gmoccapy to remove the bug. I came up with another weird thing. Sometimes the Python IDLE status is before the HALUI IDLE status.

I modified Gmoccapy.py: 
```
    def on_hal_status_interp_idle(self, widget):
        self.stat.poll()
        LOG.debug("IDLE")
        print("HAF HAF - is it realy idle?")
        print(hal.get_value("halui.program.is-idle"))
        print(self.stat.interp_state)
        print(linuxcnc.INTERP_IDLE)
```

Result without bug:
```
[Gmoccapy][DEBUG]  IDLE (gmoccapy:2576)
HAF HAF - is it realy idle?
True
1
1
```

Result with bug:
```
[Gmoccapy][DEBUG]  IDLE (gmoccapy:2576)
HAF HAF - is it realy idle?
False
1
1
```

I think there is a bug in the definition of stat.interp_state. Can anyone advise where to look for the stat.interp_state source for python?

gmoccapy, 2023-06-02:
Why do you think the interprate state is the problem?The output of both messages do differ only by the halui Hal pin, the interpreter state are the same in both outputs.So if you want to try to go deaper, I would suggest to look in the halui Hal pin stuff.IMHO this "behavior" is caused due to a race conflict.Norbert-------- Ursprüngliche Nachricht --------Von: zz912 ***@***.***>Datum: Mi., 31. Mai 2023, 20:06An: LinuxCNC/linuxcnc ***@***.***>Cc: Norbert Schechner ***@***.***>, Comment ***@***.***>Betreff: Re: [LinuxCNC/linuxcnc] Gmoccapy MSG: Must be in MDI mode..... (Issue #2453)
I tried to modify Gmoccapy to remove the bug. I came up with another weird thing. Sometimes the Python IDLE status is before the HALUI IDLE status.
I modified Gmoccapy.py:
    def on_hal_status_interp_idle(self, widget):
        self.stat.poll()
        LOG.debug("IDLE")
        print("HAF HAF - is it realy idle?")
        print(hal.get_value("halui.program.is-idle"))
        print(self.stat.interp_state)
        print(linuxcnc.INTERP_IDLE)

Result without bug:
[Gmoccapy][DEBUG]  IDLE (gmoccapy:2576)
HAF HAF - is it realy idle?
True
1
1

Result with bug:
[Gmoccapy][DEBUG]  IDLE (gmoccapy:2576)
HAF HAF - is it realy idle?
False
1
1

I think there is a bug in the definition of stat.interp_state. Can anyone advise where to look for the stat.interp_state source for python?

zz912, 2023-06-02:
Thank you Norbert for your response.

> Why do you think the interprate state is the problem?

I know the interpreter status is related to this error.
I don't know if the state of the interpreter is the result or the cause of this error.
I currently assume that the cause of this error is the state of the interpreter. I am currently trying to confirm or disprove this assumption.

I made another attempt. I added these lines to gmooccapy.py.
```
    def _periodic(self):
        # we put the poll command in a try, so if the linuxcnc pid is killed
        # from an external command, we also quit the GUI
        try:
            self.stat.poll()
        except:
            raise SystemExit("gmoccapy can not poll linuxcnc status any more")

        if hal.get_value("halui.program.is-idle") == False or self.stat.interp_state !=1:
            print("halui.program.is-idle: %s" % hal.get_value("halui.program.is-idle"))
            print("self.stat.interp_state: %i" % self.stat.interp_state)
```

Result 1: (Look at the last line)
```
halui.program.is-idle: False
self.stat.interp_state: 2
halui.program.is-idle: False
self.stat.interp_state: 2
............
............
halui.program.is-idle: False
self.stat.interp_state: 2
halui.program.is-idle: False
self.stat.interp_state: 2
halui.program.is-idle: False
self.stat.interp_state: 1
```

Result 2:
```
halui.program.is-idle: False
self.stat.interp_state: 2
halui.program.is-idle: False
self.stat.interp_state: 2
............
............
halui.program.is-idle: False
self.stat.interp_state: 2
halui.program.is-idle: False
self.stat.interp_state: 2
halui.program.is-idle: False
self.stat.interp_state: 2
```
These results were generated during the MDI_COMMAND run.

The results of this test can be explained by two theories.
Theory 1:
self.stat.interp_state works fine and halui.program.is-idle is just slower and I'm on the wrong track.
Theory 2:
halui.program.is-idle works correctly and self.stat.interp_state switches to IDLE earlier than it should.

My guess is that theory 2 is the correct one. I don't have proof for it, but a lot of other experiments suggest that it might be.

>So if you want to try to go deaper, I would suggest to look in the halui Hal pin stuff.

I have already studied this:

Source code for MDI_COMMAND finish is here: 
https://github.com/LinuxCNC/linuxcnc/blob/7baecb67c09951fc34965c53f041e9ff26d87541/src/emc/task/emctaskmain.cc#L683-L699

Source code for halui IDLE is here: 
https://github.com/LinuxCNC/linuxcnc/blob/7baecb67c09951fc34965c53f041e9ff26d87541/src/emc/usr_intf/halui.cc#L2177

But now I would like to find the Source code for Python IDLE. Unfortunately I don't know where to look.

zz912, 2023-06-11:
I bought a new, more powerful computer
DELL 9010 SFF: INTEL i5/ 16GB/ SSD 240GB/
to eliminate this problem.
I installed bookworm on it.
Unfortunately, the problem still persists.

It is strange that you are not able to simulate the problem. On the other hand, I believe you, because this bug was hiding from me for two Sundays. He is insidious.

I have been working on this issue since Apr 27. Is there anything that could motivate you developers to fix the bug?

Sigma1912, 2023-06-14:
I can reproduce this on a 2.10pre build from march.
What I have noticed is that sometimes while in the jogging screen and changing  the mdi-command-xx pins in halshow  it will switch to MDI screen but not go back to the joggin screen as it usually does.

zz912, 2023-06-14:
Hello everybody,

I spent this evening again looking for the source of this bug.

I am convinced that the source of this bug is the bad functionality of EMC_TASK_INTERP. I spent tonight to prove that EMC_TASK_INTERP works badly.

I would like to ask you to confirm or refute my theory.

Since I don't want to waste your precious time, I have prepared an attempt that has no dependence on previous posts. If you are willing to help me, just read only this post.

In this part of the Gmoocapy code
https://github.com/LinuxCNC/linuxcnc/blob/67b8140c2788073cf9f95e4bef67271b1ea53e0c/src/emc/usr_intf/gmoccapy/gmoccapy.py#L2572-L2622
I would assume that the commands will be executed when the LCNC is only in IDLE.
QUESTION 1:
Is my assumption correct?

To verify that LCNC is in IDLE, I added the following lines to Gmoccapy's source code:
```
   def on_hal_status_interp_idle(self, widget):
        print("HAF HAF - linuxcnc.INTERP_IDLE = " + str(linuxcnc.INTERP_IDLE))
        print("HAF HAF - linuxcnc.INTERP_READING = " + str(linuxcnc.INTERP_READING))
        print("HAF HAF - linuxcnc.INTERP_PAUSED = " + str(linuxcnc.INTERP_PAUSED))
        print("HAF HAF - linuxcnc.INTERP_WAITING = " + str(linuxcnc.INTERP_WAITING))
        print("HAF HAF - now is " + str(self.stat.interp_state))
```
I would expect self.stat.interp_state to be equal to linuxcnc.INTERP_IDLE.
QUESTION 2:
Is my expectation correct?

Now let's watch the video:
[image]

When we look at the video, we find out that mostly LCNC is READING (2) and at the end of the video, LCNC is IDLE (1).
QUESTION 3:
Can LCNC READING (2) be considered a bug?
QUESTION 4:
Can the random status of LCNC IDLE (1) and LCNC READING (2) be considered a bug?

This video shows mostly state 2 and once 1. During longer testing, states 1 and 2 appear more randomly.

Please do not take this post of mine as offensive or sarcastic. I really just need help. I am very unhappy with this bug.

I asked the questions in such a way that they could be answered simply yes/no and did not delay you.
QUESTION 1: yes/no
QUESTION 2: yes/no
QUESTION 3: yes/no
QUESTION 4: yes/no

phillc54, 2023-06-15:
Do you need to poll the status channel to ensure that it is up to date before reading it?
`self.stat.poll()`

zz912, 2023-06-15:
> Do you need to poll the status channel to ensure that it is up to date before reading it?
> `self.stat.poll()`

Oh yeah, you're right. I'm an idiot :-(.

```
3 2
HAF HAF - self.stat.poll() was executed
HAF HAF - linuxcnc.INTERP_IDLE = 1
HAF HAF - linuxcnc.INTERP_READING = 2
HAF HAF - linuxcnc.INTERP_PAUSED = 3
HAF HAF - linuxcnc.INTERP_WAITING = 4
HAF HAF - now is 1
Must be in MDI mode to issue MDI command
3 2
Must be in MDI mode to issue MDI command
HAF HAF - self.stat.poll() was executed
HAF HAF - linuxcnc.INTERP_IDLE = 1
HAF HAF - linuxcnc.INTERP_READING = 2
HAF HAF - linuxcnc.INTERP_PAUSED = 3
HAF HAF - linuxcnc.INTERP_WAITING = 4
HAF HAF - now is 1
3 2
3 2
HAF HAF - self.stat.poll() was executed
HAF HAF - linuxcnc.INTERP_IDLE = 1
HAF HAF - linuxcnc.INTERP_READING = 2
HAF HAF - linuxcnc.INTERP_PAUSED = 3
HAF HAF - linuxcnc.INTERP_WAITING = 4
HAF HAF - now is 1
3 2
HAF HAF - self.stat.poll() was executed
HAF HAF - linuxcnc.INTERP_IDLE = 1
HAF HAF - linuxcnc.INTERP_READING = 2
HAF HAF - linuxcnc.INTERP_PAUSED = 3
HAF HAF - linuxcnc.INTERP_WAITING = 4
HAF HAF - now is 1

```

zz912, 2023-06-15:
[Sigma1912](https://github.com/Sigma1912)
> I can reproduce this on a 2.10pre build from march. This is a simulation machine without RT-kernel. What I have noticed is that sometimes while in the jogging screen and changing the mdi-command-xx pins in halshow it will switch to MDI screen but not go back to the joggin screen as it usually does.

This bug is possibly the cause of other problems. If it's not resolved, we can't move on. For example, when this error occurs, nonsensical tool corrections are displayed.
[image]

That's why I'm sorry that there is no interest in fixing this bug because it makes using LCNC very dangerous when using ATC.

Sigma1912, 2023-06-15:
AXIS gui does not seem to have this bug, would that not indicate that the problem is with gmoccpy rather than with python stat?

zz912, 2023-06-15:
> AXIS gui does not seem to have this bug,

This bug is really insidious, the fact that it did not show up in AXIS does not mean that it is not there.
I'm paranoid. I've been looking for that bug for a long time.

Assuming this bug is not in AXIS, we can rule out halui.

I think AXIS doesn't use python stat. It is so?
That's why I didn't rule out python stat.

phillc54, 2023-06-15:
AXIS does use linuxcnc.stat
[L3540](https://github.com/LinuxCNC/linuxcnc/blob/9f65a9903b1cb5af18ac0d9d328a49ff83ac18a1/src/emc/usr_intf/axis/scripts/axis.py#L3540)

[13 more replies]

closed 2023-10-08 by gmoccapy