LinuxCNC/linuxcnc #2466: Pncconf error message when choosing less than 3 step generators

YClayton, 2023-05-03:
## Here are the steps I follow to reproduce the issue:

 1. Create a new setup.
 2. Click through to the Mesa Card 0 page.
 3. Select 7i76e - internal data from the list.
 4. Select Number of Step Generator = 1
 5. Click Accept Components Changes.
 6. Select I/O Connector2 (or TB2/TB3 in the release with v2.9) tab... only one step gen available, assign main axis X to it.
 7. Click the Forward button to advance to the next page.
 8. Error message appears...
 9. You forgot to designate a stepper or PWM signal for axis Y  <and> You forgot to designate a stepper or PWM signal for axis Z.
 10. You can't move on until you reconfigure for 3 to 5 step gens.

## This is what I expected to happen:

Configure 1 step gen and continue

## This is what happened instead:

Error Message

## Information about my hardware and software:

I had the same issue on v2.8.4 and v2.9

andypugh, 2023-05-04:
Sorry Chris, I am sure you are busy, but this this looks like one for you.

c-morley, 2023-05-04:
It's an understandable error if you picked an XYZ machine but only designated one motor drive signal.
What are you trying to configure?

YClayton, 2023-05-04:
Perhaps an error message that says something to the effect...

'The number of step generators you have chosen does not match the selected **Axis Configuration** on the previous **Base Information** page.'

c-morley, 2023-05-04:
The message now tells u exactly what's wrong. If one wants to add more info about why it sure cant hurt. As far as adding more machine options sure, but pncconf has run out of developer enthusiasm, so I doubt it will happen.

YClayton, 2023-05-04:
Understood. Thanks for pointing out my error.

c-morley, 2023-05-04:
I might have been a bit too pointed with my comment..sorry. we do occasionally upgrade pnc and I will try to remember to improve the message. Please keep critiques and issues coming. :)

YClayton, 2023-05-04:
Thanks.
 My goal is to be of help, not a pita. Being new to Linux, and LinuxCNC, I have a fresh set of eyes looking at things. I hate to simply complain about thing with being proactive and offering some sort of ideas or a solution to fix the issue that I ran into.

I also understand that some things can go much faster and smoother without the "help" from folks that are too ignorant on the subject.

In the meantime I'll see about creating a pull request that updates the documentation regarding the relationship between machine type and step generators.

phillc54, 2023-05-05:
You also need to be aware that it is possible to have more than one stepgen per axis. This is determined when/if you add a tandem axis during stepgen selection.