LinuxCNC/linuxcnc #2467: Strange NML estop issue

rene-dev, 2023-05-04:
checkout master
comment out emcLubeOff() here: https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/task/emctask.cc#L346
run sim/axis
9 out of 10 times you cannot get the machine out of estop.
also, many tests fail as exiting estop doesnt work reliably, for example abort/feed-rate
if you replace emcLubeOff() with emcCoolantFloodOff() its working again.

rene-dev, 2023-05-04:
emcAuxEstopOff() sends EMC_AUX_ESTOP_OFF using forceCommand,
emcIoAbort() sends EMC_TOOL_ABORT using sendCommand.

adding a printf in io shows only EMC_AUX_ESTOP_OFF_TYPE or EMC_TOOL_ABORT_TYPE is ever received in io.
if only EMC_TOOL_ABORT_TYPE is reveived, the machine does not exit estop.

probably the best solution is to remove EMC_TOOL_ABORT_TYPE, as it is not called anyway if the machine does get out of estop.

rene-dev, 2023-05-04:
turns out both commands are force-queued.
just removing emcIoAbort doesnt fix it reliably.
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/task/taskclass.cc#L167

c-morley, 2023-05-05:
Is it used in iov2?

rene-dev, 2023-05-05:
Both components do something with the message. It has nothing to do with v2

rene-dev, 2023-05-06:
probably also the cause of #139

closed 2023-09-26 by rene-dev