LinuxCNC/linuxcnc #2468: build 2.9.0-pre0

grandixximo, 2023-05-05:
## Here are the steps I follow to reproduce the issue:

 1.downlod code for 2.9.0-pre0 branch
 2.follow instructions https://linuxcnc.org/docs/html/code/building-linuxcnc.html#_building_debian_packages

## This is what I expected to happen:

Debian packages will be built

## This is what happened instead:

`SyntaxError: from __future__ imports must occur at the beginning of the file
Depending streamer.9
Depending linuxcnc.1
Depending hy_gt_vfd.1
make[1]: *** No rule to make target '../bin/halcompile', needed by '../docs/man/man1/thermistor.1'.  Stop.
make[1]: *** Waiting for unfinished jobs....
Depending halstreamer.1
make[1]: Leaving directory '/home/aitalmac/lxcnc 2.9.0-pre0/src'
make: *** [debian/rules:34: build-stamp] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
`
## It worked properly before this:
not sure

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian 10 Buster
 * I am using this kernel version (shown by `uname -a`): 4.19.0-24-rt-amd64 
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.9.0-pre0

SebKuzminsky, 2023-05-05:
The first line of your log, `SyntaxError: from __future__ imports must occur at the beginning of the file` makes me think you don't have the source that you think you do.

How did you get the source code for LinuxCNC 2.9.0-pre0?

If you got the source using git, what's the output of `git describe`?

grandixximo, 2023-05-05:
I downloaded zip of the branch from github.
I did also try get the branch with git, same result, I deleted the one I had done with git now.

SebKuzminsky, 2023-05-05:
The version number of the 2.9 branch is 2.9.0-pre1 now, not pre0.  The change from pre0 to pre1 happened on November 2022.

Is it an old zip?  If so, please try building a newer version, there have been many improvements to the python code in particular, which is where that build error seems to come from.

grandixximo, 2023-05-05:
It's a zip I download few hours ago, 2.9.0-pre1 builds fine without docs, I wanted to try 2.9.0-pre0 because I'm wanting to check some old bug, but I found cause for my bug already, just letting you guys know 2.9.0-pre0 doesn't build no more.

SebKuzminsky, 2023-05-05:
Ok, thanks for letting us know.

2.9.0~pre0 is not a supported release, all users and developers should move up to the tip of the 2.9 branch, which is supported.

closed 2023-05-05 by SebKuzminsky