LinuxCNC/linuxcnc #2482: `override_limits` stays active in homed state

hansu, 2023-05-10:
The [docs](http://linuxcnc.org/docs/2.8/html/code/code-notes.html#_override_limits) say: "The OVERRIDE_LIMITS command prevents limits from tripping **until the end of the next JOG command.** "

If the machine is not homed it works as described. 
But if the machine is homed you can jog around as long as you want with limit switch active and override enabled.
Yes I know it's a rare use case because this usually can't happen if you have set the working area correctly. But however, the hard limits should be prioritized higher than the soft limits.

### Steps to reproduce:
1. Start linuxcnc
2. home all
3. set a limit switch to 1
4. run override_limits()
5. switch the machine on again
6. jog around

### Expected:

Turn off the machine if first jogging is stopped.

### What happened instead:

Jogging still possible, machine stays on.

andypugh, 2023-09-28:
I think this is probably meant to be handled in this switch:
https://github.com/LinuxCNC/linuxcnc/blob/2.8/src/emc/motion/control.c#L1233
(note that this file is tabs-and-spaces indented, so needs to be set up right to follow the structure)
In "case EMCMOT_MOTION_FREE:" (Joint jogging) there is this line:
https://github.com/LinuxCNC/linuxcnc/blob/2.8/src/emc/motion/control.c#L1309
Which clears the override flag. 

I don't see an equivalent line in the other cases of the switch. But haven't figured out where it belongs yet.

andypugh, 2023-09-28:
Note for anyone looking at this, most of the sim configs don't connect the limit pins, so don't show the checkbox. 
sim/axis/vismach  configs VNC / HBM and 5Axis do display the box.

andypugh, 2023-09-29:
It is definitely https://github.com/LinuxCNC/linuxcnc/blob/2.8/src/emc/motion/control.c#L1309 which clears the override in joint mode. Commenting it out makes the joint mode behave like axis mode, and the checkbox remains set. 

I have tried inserting the same code, with and without the guarding condition in various places in case EMCMOT_MOTION_TELEOP:  and I can make the override clear immediately at power-on, or leave the behaviour unchanged, but I have not as yet worked out where to put it to achieve the desired effect.

mozmck, 2025-01-01:
I had this problem and the following seems to fix it for me.
In the EMCMOT_MOTION_TELEOP case statement insert these two block of code.
The first one is at the top just after this line:
```c++
        axis_sync_carte_pos_to_teleop_tp(+1, pcmd_p); // teleop
``` 
Block 1
```c++
        if ( axis_jog_is_active() ) {
            /* is any limit disabled for this move? */
            if ( emcmotStatus->overrideLimitMask ) {
                emcmotInternal->overriding = 1;
            }
        }
```
The second block is just before the break for the EMCMOT_MOTION_TELEOP case statement:
```c++
	/* END OF OUTPUT KINS */

	/* if overriding is true and the jog is complete, the limits should be re-enabled */
        if ( ( emcmotInternal->overriding ) && ( !axis_jog_is_active() ) ) {
            emcmotStatus->overrideLimitMask = 0;
            emcmotInternal->overriding = 0;
        }

	/* end of teleop mode */
```

closed 2025-01-04 by andypugh