LinuxCNC/linuxcnc #2489: Combination M61 + G4

zz912, 2023-05-11:
I use RIP instalation Branche 2.9, sim gmoccapy

If I enter a combination of M61 and G4 commands in the command line, sometimes the G4 command is executed, sometimes not.

Watch the video:

https://github.com/LinuxCNC/linuxcnc/assets/96618597/b8a00f0e-d288-4b07-a1a1-aa126b025976

zz912, 2023-05-12:
1) If you cannot view the video, please download it and play it in VLC player. On some PCs the video plays, on some PCs the video says that it is broken.

2) I think this bug is the cause of these bugs:
https://github.com/LinuxCNC/linuxcnc/issues/2453
https://github.com/LinuxCNC/linuxcnc/issues/2448
I tried to look for the cause of these bugs and found that the execution of MDI_COMMANDS is random. Maybe this bug isn't the cause, but I can't look any further.

3) I managed to simulate the bug:
- on a physical PC, RIP installation, branche 2.9
- on Oracle VM VirtualBox, RIP installation, branche 2.9 (reduce CPU is irrelevant)
- on Oracle VM VirtualBox, Debian installation, branche 2.8.4 (CPU reduction is irrelevant)

4) I have a problem that HansU wants to help me solve the previous bug, but he can't simulate it. Therefore, I would like to ask for feedback if this error also appears.

zz912, 2023-05-14:
I made tests of this bug:
2.8.0~pre3, GUI AXIS, BUG - NO
2.8.0~pre3, GUI GMOCCAPY, BUG - YES
2.8.0~pre3, GUI GSCREEN, BUG - NO
2.8.0~pre3, GUI SILVERDRAGON, BUG - NO

2.7.15,  GUI AXIS, BUG - NO
2.7.15,  GUI GMOCCAPY, BUG - NO

zz912, 2023-07-16:
The cause of this problem is in these lines:
https://github.com/LinuxCNC/linuxcnc/blob/56883b969838b721c0b0ebe12b9c14abf91d640e/src/emc/usr_intf/gmoccapy/gmoccapy.py#L3540-L3544

As these lines cause more problem, I made a HAL pin which disable these lines.
https://github.com/LinuxCNC/linuxcnc/pull/2571

Here is the test result:
[image]

When I change tools Q1 to Q5, the lines are activated. This is the default behavior of Gmoccapy. Note that LCNC ignores G4 P10.

When I change tools Q6 to Q10, the lines are disabled by my HAL pin. LCNC is working properly.

gmoccapy, 2023-07-27:
I will be on this point in early September and double check all code and hope to find a solution for all the issues you mentioned.

Could you please do a test for me?
Disable the lines in "tool in spindle changed" and then set the retain G43 parameter in the INI.
Have the bug gone away?

Thanks for you help

zz912, 2023-07-28:
I will do the test and then I will give you feedback.

zz912, 2023-07-31:
I tried to do that test, but I couldn't get RETAIN_G43 to work:
https://forum.linuxcnc.org/38-general-linuxcnc-questions/49657-retain-g43

zz912, 2025-01-08:
Fixed in https://github.com/LinuxCNC/linuxcnc/pull/3113

closed 2025-01-08 by zz912