LinuxCNC/linuxcnc #2498: PNCCONF Fails to enter MESA page when reloading existing configuration

PapaGolf0422, 2023-05-21:
Running Axis v2.10.0-pre0-1396 on Debian 4.19.0-23-rt-amd64 #1 SMP PREEMPT RT Debian 4.19.269-1 (2022-12-20) x86_64 GNU/Linux

I created a profile using PNCConf for a lathe using a MESA 5i25 PCI card with a custom bitfile using the discover card function.
When I reopened the config using PNCConf, I hit next until I should have arrived at the Mesa Config page but instead recieved an error.  Screenshot attached as it wouldnt let me copy the error text.

I created a test profile with a stock 5i25 config and everything worked as it should.

[image]

c-morley, 2023-06-11:
I believe this is fixed now.

VladRTT, 2023-10-08:
Absolutely same error. How can I fix it? Please, help!!

c-morley, 2023-10-08:
Not sure how you got linuxcnc but you should upgrade it to the latest.

VladRTT, 2023-10-08:
> Not sure how you got linuxcnc but you should upgrade it to the latest.

I have installed it from [here ](https://linuxcnc.org/docs/devel/html/getting-started/getting-linuxcnc.html)
Now, I'm trying to build it from src ([instruction](https://linuxcnc.org/docs/devel/html/code/building-linuxcnc.html))

VladRTT, 2023-10-08:
It's fail. I have reinstall from synaptic manager, and I have this error again.
[image]

c-morley, 2023-10-08:
Still seems you have an older version.
The official Debian repository is extremely out of date unfortunately.
using a buildbot version or building from source will give you the latest.

VladRTT, 2023-10-08:
I looked for buildbot dists and haven't found bookworm dist.
[image]

phillc54, 2023-10-08:
There is another buildbot [here](http://buildbot2.highlab.com/)