LinuxCNC/linuxcnc #2508: emctaskmain.cc - todo FIXME

zz912, 2023-05-25:
I found a lot of "todo FIXME" notes in emctaskmain.cc. I would like to ask someone more experienced to look into it.

https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/task/emctaskmain.cc#L518
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/task/emctaskmain.cc#L529
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/task/emctaskmain.cc#L1082
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/task/emctaskmain.cc#L2137
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/task/emctaskmain.cc#L2565
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/task/emctaskmain.cc#L3477

andypugh, 2023-06-10:
I wonder how long those have been there for? 
MGS could be the initials of the commenter.

zz912, 2023-06-11:
> I wonder how long those have been there for?

First FIXME 
https://github.com/LinuxCNC/linuxcnc/blob/beacb3c0572eee72d0faf80cb736610059c37768/src/emc/task/emctaskmain.cc#L518
jepler committed on Mar 18, 2010
https://github.com/LinuxCNC/linuxcnc/commit/b1689654996c0297ec56478a9f2460d9e62fe4f8