LinuxCNC/linuxcnc #2509: linuxcnc does not build with clang16

rene-dev, 2023-05-26:
it fails with the following error with clang from https://apt.llvm.org/
> Linking ../rtlib/abs.so
ld -d -r -o objects/abs.tmp objects/rtobjects/hal/components/abs.o
objcopy -j .rtapi_export -O binary objects/abs.tmp objects/abs.sym
(echo '{ global : '; tr -s '\0' < objects/abs.sym | xargs -r0 printf '%s;\n' | grep .; echo 'local : * ; };') > objects/abs.ver
grep: (standard input): binary file matches
clang-16 -shared -Bsymbolic -L/home/rene/dev/linuxcnc-stock/lib -Wl,-rpath,/home/rene/dev/linuxcnc-stock/lib -ltirpc   -Wl,--version-script,objects/abs.ver -o ../rtlib/abs.so objects/rtobjects/hal/components/abs.o -lm
/usr/bin/ld:objects/abs.ver:13: ignoring invalid character `\017' in script
/usr/bin/ld:objects/abs.ver:13: ignoring invalid character `\037' in script
/usr/bin/ld:objects/abs.ver:13: syntax error in VERSION script
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [Makefile:1234: ../rtlib/abs.so] Error 1

rene-dev, 2023-05-26:
the problem is here: https://github.com/LinuxCNC/linuxcnc/blob/master/src/Makefile#L1233
for some reason the symbols include invalid characters. I dont know if only clang generates them, or if only clang doesnt like them in version script.

rene-dev, 2023-05-26:
I tried changing this to -export-symbols-regex, it seems to be the more portable and better way of doing this.
https://www.gnu.org/software/gnulib/manual/html_node/LD-Version-Scripts.html
but doesnt work for all modules yet.

NTULINUX, 2023-05-27:
Rene, thanks for working on this, means a lot! Can you post the exact export-symbols-regex modification you tried and I'll try to get it working with the other modules? I'd love to help!