LinuxCNC/linuxcnc #2515: Update "Updating LinuxCNC" for 2.9 release

hansu, 2023-06-04:
The docs page [Updating LinuxCNC](http://linuxcnc.org/docs/2.9/html/getting-started/updating-linuxcnc.html) needs to be updated for 2.9.

andypugh, 2023-06-18:
Updating this is part of the Releasing Process.

hansu, 2023-07-20:
This should be mentioned: https://forum.linuxcnc.org/gmoccapy/49580-pyngcgui-error#275835

Not sure if it's a good idea to replace that file automatically by the update-ini script and I don't know if that is even possible.

andypugh, 2023-07-24:
The update script only knows how to change values in INI and HAL files. 

From reading the thread it sounds like one thing that needs  to be done is to update the versions of some files in the tree if some of the sim configs are not working. 
(We probably should only have one version of these files, and fix it in one place, rather than having several of them spread through the configs tree) 

However, this is not a fix for cases where the UI is in the user's config folder (as is probably the usual case). We could change the file path in the INI to point to a library version, or insert a comment in the INI to give them an indication of the fix. Or possibly pop-up a message box.

hansu, 2023-08-03:
I think the best option would be to pop-up a message box and point to the library version so that the user can replace the file in the his config folder.

zz912, 2023-10-02:
Hello,

I would like to ask you to add the problem with Retain G43 to the TODO list for 2.9 relase.
https://forum.linuxcnc.org/38-general-linuxcnc-questions/49657-retain-g43

I didn't put this issue on Github because Norbert put it in his TODO list.
https://forum.linuxcnc.org/38-general-linuxcnc-questions/49657-retain-g43#276648

If RETAIN_G43 could be fixed, it would be possible to delete the code parts that did this from Gmoccapy.
Deleting these parts of the codes would solve a few random bugs on the M6 Tx by itself.

To keep the same behavior from older versions, it will be necessary to check that PncConf for Gmoccapy generates RETAIN_G43 = 1 in INI

andypugh, 2023-10-04:
Doc Updated
(With regards to the things that Norbert plans to do, expect those in 2.9.1)

closed 2023-10-04 by andypugh