LinuxCNC/linuxcnc #2516: Bug in rtapi_shmem_new

zkozelr, 2023-06-06:
## Here are the steps I follow to reproduce the issue:
 1. read the code in uspace_common.h (around line 73)

## This is what I expected to happen:
Find an existing shmem segment in the shmem_array even if there are free slots in the array (from previously deallocated segments)

## This is what happened instead:
The iteration (search) will abort on the first free slot instead of searching through all allocated slots which will result in creating a duplicate shmem_array entry for a segment. At minimum, the ref count within this entry will then be wrong. I am not sure what other consequences the duplication may have.

## Information about my hardware and software:
Found by studying code in the 2.9 branch of linuxcnc

SebKuzminsky, 2023-06-07:
Hi @zkozelr, thanks for the bug report.

https://github.com/LinuxCNC/linuxcnc/blob/2.9/src/rtapi/uspace_common.h#L60-L80

One sequence of events that would trigger the bug looks like this:

1. Load the first realtime component.  Its setup function calls `rtapi_shmem_new()` with key A.  `rtapi_shmem_new()` searches `shmem_array` and finds an uninitialized `shmem_array` entry (`.magic != SHMEM_MAGIC`) at index 0.  It initializes the entry and returns shmem handle 0 for key A.
2. The component then calls `rtapi_shmem_new()` again, with key B.  A similar search as in (1) returns shmem handle 1 for key B.
3. The component then calls `rtapi_shmem_delete()` with shmem handle 0 (corresponding to key A).  The refcount drops to 0 and the `shmem_array` entry is de-initialized, specifically `.magic` is set to 0.
4. Load a second realtime component.  Its setup function calls `rtapi_shmem_new()` with key B.  The test on line 69 fails, and the test on line 73 fails, and `rtapi_shmem_new()` incorrectly creates a second `shmem_array` entry for key B at index 0.

This looks like a real bug to me: having multiple `shmem_array` entries for a single underlying shared memory segment is clearly bogus and not what the intent of the code is.

But, amusingly, it looks like a toothless bug, at least on uspace: even if it were to happen, i don't think there would be any perceptible damage done.  The two realtime components in the example above would use different pointers and refcounts for the same underlying shared memory, but the underlying shared memory would be the same, and would survive until all the `shmem_array` entries referencing it were all deleted.  

I like @zkozelr's suggestion to change the search loop (lines 68-74) to search the entire `shmem_array` for the requested key, not just the part of the array before the first uninitialized entry.  If the key is not found, the function could pick any of the uninitialized entries to hold the new shmem segment.

@zkozelr, does that change seem right to you?  Do you want to implement this or should I?

@andypugh, do you want this fix in 2.8?  If not i think we should put it in 2.9.

zkozelr, 2023-06-07:
@SebKuzminsky I do have a patch and as soon as I figure out how to get past  "ERROR: Permission to LinuxCNC/linuxcnc.git denied to zkozelr." (while trying to push my branch) I'll create a PR.

andypugh, 2023-06-07:
> @andypugh, do you want this fix in 2.8? If not i think we should put it in 2.9.

2.9, I think. If the bug hasn't show up.
I think it would be an unusual case to even have multiple shmem areas? Does hal_streamer create a new shmem? I have written experimental components that create an extra private shmem (for laser rastering) but those were neve committed.

andypugh, 2023-06-07:
> I do philosophically oppose creating useless forks, so please feel free to tell me what you'd like to see happen. 

Typically the process that Github is set up for is to create a useless fork and then delete it....

SebKuzminsky, 2023-06-07:
> @zkozelr 
> I do philosophically oppose creating useless forks, so please feel free to tell me what you'd like to see happen.
> [issue_2516.txt](https://github.com/LinuxCNC/linuxcnc/files/11672974/issue_2516.txt)

One advantage of "creating a useless fork" and putting your patch there is that we can use the github infrastructure to view the patch and talk about it, and the github CI infrastructure will automatically test it for us.  So it's not actually useless.

I downloaded your patch and looked at it, and it looks right to me.  I applied it to a new branch and pushed it and opened a PR here: #2518.

> @zkozelr 
> Why can't I enclose a .patch file?

One annoying thing about github is that it restricts the extensions of the files it lets you upload/attach.

<https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/attaching-files>

Your trick of just renaming it to .txt  was a good call.

> @andypugh 
> I think it would be an unusual case to even have multiple shmem areas? Does hal_streamer create a new shmem?

`git grep rtapi_shmem_new` in `linuxcnc/src` shows 15 non-test, non-comment instances, I think.

HAL makes one, each hal_stream_t makes one, uspace rtapi makes a shmem of its own, there's one for communication between Motion and non-realtime (cmd/status/etc), one in ClassicLadder, one in halscope, etc.