LinuxCNC/linuxcnc #2530: G95 feed per rev not working when using 2nd spindle

hdpreez, 2023-06-19:
When using the 2nd spindle on a multiple spindle lathe setup, the g95 feed per revolution does not work. It will only feed in g95 mode when the main (spindle.0) is running

## Here are the steps I follow to reproduce the issue:
Use the  sim.axis.lathe_multispindle.. Home and excecute the following in MDI

g21 g0 x50
g96 s140 d5000 $1
m3$1
g95 f0.2
g1 x0

## This is what I expected to happen:
The feed is expected to start and increase according to the spindle speed (G96 CSS mode) at a rate of 0.2mm per revolution

## This is what happened instead:
Nothing, it is as if the spindle isnt running, error report on Axis states that spindle is not running. Trying g95 $1  f0.2 results in error "$ (spindle selection) word with no G41, G41.1, G42, G42.1, G76, G95 or G96 to use it"

## It worked properly before this:
Not tested, multi spindle only available in 2.8

## Information about my hardware and software:

 * Running Debian Buster, release 2.8.3 and also tested with 2.8.4
 * Mesa ethernet FPGA +7i77 + io cards
 * Checked that spindle.1.speed-in, spindle.1.revs and spindle.1.at-speed is configured properly. 
 
Henk

andypugh, 2023-07-08:
This should be fixed by c15ac50957f341a0ed35d8aa2b2757f108816104
Sorry for the inconvenience. 
To actually get the fix working on your machine you would have to build from source, or wait for the buildbot to complete building the fixed version and install from there. Instructions on how to do so are at http://buildbot.linuxcnc.org
I am going to close this issue, but feel free to comment here if you need any help with getting the fix running for you.

closed 2023-07-08 by andypugh

hdpreez, 2023-07-10:
Thanks, will wait for the next update or 2.9 release,  workaround as posted[]( https://forum.linuxcnc.org/49-basic-configuration/49340-g95-feed-not-working-with-2nd-spindle?start=0
) works for now

I used a mux4 to channel the vel-fb from each spindle to spindle.0.speed-in depending on which spindle is running by linking spindle.0.on and spindle.1.on to mux4.0.sel0 and sel1 respectively. 

spindle.0.speed-in = spindle 0 vel fb when spindle 0 is running and spindle 1 is off
spindle.0.speed-in = spindle 1 vel fb when spindle 1 is running and spindle 0 is off
spindle.0.speed-in = spindle 0 vel fb when both spindles are running