LinuxCNC/linuxcnc #2538: Dependencies problem while building from source

nomi-ramzan, 2023-06-22:
## Here are the steps I follow to reproduce the issue:

 1. Install Fedora IOT and clone linuxcnc repo using git [this link](https://linuxcnc.org/docs/html/code/building-linuxcnc.html)
 2. Install dependencies ie `tcl-devel, libtirpc-devel, libusb1-devel, intltool, boost-devel ...`
 3. autoreconf --install
 4. ./autogen.sh
 5. ./configure --with-realtime=uspace

## This is what I expected to happen:
Successfully build linuxcnc for Fedora
## This is what happened instead:

```
checking for BWidget using /usr/bin/tclsh8.6... found
checking for BLT using /usr/bin/tclsh8.6... not found
checking for tclX using /usr/bin/tclsh8.6... no
configure: error: Tclx not found!
    install with "sudo apt-get install tclx"
[root@localhost src]# sudo apt-get install tclx
Reading package lists... Done
Building dependency tree... Done
E: Unable to locate package tclx

```

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `cat /etc/`): 
   ```
    NAME="Fedora Linux"
    VERSION="38.20230619.0 (IoT Edition)"
    ID=fedora
    VERSION_ID=38
    VERSION_CODENAME=""
    PLATFORM_ID="platform:f38"
    PRETTY_NAME="Fedora Linux 38.20230619.0 (IoT Edition)"
    ANSI_COLOR="0;38;2;60;110;180"
    LOGO=fedora-logo-icon
    CPE_NAME="cpe:/o:fedoraproject:fedora:38"
    HOME_URL="https://fedoraproject.org/"
    DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f38/system-administrators-guide/"
    SUPPORT_URL="https://ask.fedoraproject.org/"
    BUG_REPORT_URL="https://bugzilla.redhat.com/"
    REDHAT_BUGZILLA_PRODUCT="Fedora"
    REDHAT_BUGZILLA_PRODUCT_VERSION=38
    REDHAT_SUPPORT_PRODUCT="Fedora"
    REDHAT_SUPPORT_PRODUCT_VERSION=38
    SUPPORT_END=2024-05-14
    VARIANT="IoT Edition"
    VARIANT_ID=iot
    OSTREE_VERSION='38.20230619.0'
```
 * I am using this kernel version (shown by `uname -a`): Linux localhost.localdomain 6.3.8-200.fc38.aarch64 SMP PREEMPT_DYNAMIC Thu Jun 15 02:15:18 UTC 2023 aarch64 GNU/Linux
 * I am running ...
   - [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   - [x] A binary I built myself
   - [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.10.0-pre0-1542-g93fccc095b
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): GPIO on raspberry pi 4b 8GB

petterreinholdtsen, 2023-06-22:
[Nauman]
>  2. Install dependencies ie `tcl-devel, libtirpc-devel, libusb1-devel,
>  intltool, boost-devel ...`

I guess this list was incomplete.

> checking for BLT using /usr/bin/tclsh8.6... not found
> checking for tclX using /usr/bin/tclsh8.6... no
> configure: error: Tclx not found!
>     install with "sudo apt-get install tclx"
> ***@***.*** src]# sudo apt-get install tclx

Any idea which Fedora package provide the tclx stuff?

I use Debian myself, and have no idea about Fedora package names.

-- 
Happy hacking
Petter Reinholdtsen

SebKuzminsky, 2023-06-22:
The `src/configure` script looks for runtime dependencies as well as build dependencies.  Install them or configure with `--disable-check-runtime-deps`.

closed 2023-06-22 by SebKuzminsky

nomi-ramzan, 2023-06-24:
> [Nauman]
> 2. Install dependencies ie `tcl-devel, libtirpc-devel, libusb1-devel, intltool, boost-devel ...`
> I guess this list was incomplete.
> checking for BLT using /usr/bin/tclsh8.6... not found [...]

I found the solution. `install with "sudo dnf install tclx"`

nomi-ramzan, 2023-06-24:
I complete build process in Fedora but didn't find linuxcnc command. where should I find this ?

zz912, 2023-07-01:
./linuxcnc-2.9/scripts/linuxcnc

SebKuzminsky, 2023-07-01:
> I complete build process in Fedora but didn't find linuxcnc command. where should I find this ?

Your original message shows that you are building for run-in-place.  Per the build instructions at <http://linuxcnc.org/docs/devel/html/code/building-linuxcnc.html#Quick-Start>, you next need to run the `rip-environment` script.  It will add your just-built LinuxCNC executables to your shell's PATH.