LinuxCNC/linuxcnc #2543: Build linuxcnc on fedora 37

nomi-ramzan, 2023-06-26:
## Here are the steps I follow to reproduce the issue:

 1. Install Fedora IOT and clone linuxcnc repo using git [this link](https://linuxcnc.org/docs/html/code/building-linuxcnc.html)
 2. Install dependencies ie `tcl-devel, libtirpc-devel, libusb1-devel, intltool, boost-devel ...` [full list](https://stackoverflow.com/questions/76555708/build-linuxcnc-on-fedora-37)
 3. autoreconf --install
 4. ./autogen.sh
 5. ./configure --with-realtime=uspace

## This is what I expected to happen:
Successfully build linuxcnc for Fedora
## This is what happened instead:

```
checking for readline in -lreadline... yes
Trying readline with -ltermcap
checking for readline in -lreadline... yes
Readline does not require -ltermcap
checking whether readline license is compatible with GPL-2... no
configure: WARNING: The LinuxCNC binary you are building may not be
distributable due to a license incompatibility with LinuxCNC (some portions
GPL-2 only) and Readline version 6 and greater (GPL-3 or later).
configure: error: To configure LinuxCNC in this way, you must
invoke configure with "--enable-non-distributable=yes".  Note that on
Debian-based systems, you may be able to use libreadline-gplv2-dev instead.

```

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `cat /etc/`): 
   ```
    NAME="Fedora Linux"
    VERSION="38.20230619.0 (IoT Edition)"
    ID=fedora
    VERSION_ID=38
    VERSION_CODENAME=""
    PLATFORM_ID="platform:f38"
    PRETTY_NAME="Fedora Linux 38.20230619.0 (IoT Edition)"
    ANSI_COLOR="0;38;2;60;110;180"
    LOGO=fedora-logo-icon
    CPE_NAME="cpe:/o:fedoraproject:fedora:38"
    HOME_URL="https://fedoraproject.org/"
    DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f38/system-administrators-guide/"
    SUPPORT_URL="https://ask.fedoraproject.org/"
    BUG_REPORT_URL="https://bugzilla.redhat.com/"
    REDHAT_BUGZILLA_PRODUCT="Fedora"
    REDHAT_BUGZILLA_PRODUCT_VERSION=38
    REDHAT_SUPPORT_PRODUCT="Fedora"
    REDHAT_SUPPORT_PRODUCT_VERSION=38
    SUPPORT_END=2024-05-14
    VARIANT="IoT Edition"
    VARIANT_ID=iot
    OSTREE_VERSION='38.20230619.0'
```
 * I am using this kernel version (shown by `uname -a`): Linux localhost.localdomain 6.3.8-200.fc38.aarch64 SMP PREEMPT_DYNAMIC Thu Jun 15 02:15:18 UTC 2023 aarch64 GNU/Linux
 * I am running ...
   - [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   - [x] A binary I built myself
   - [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.10.0-pre0-1542-g93fccc095b
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): GPIO on raspberry pi 4b 8GB

rmu75, 2023-06-26:
Look at the last 2 lines, it tells the problem and what to do.

rene-dev, 2023-06-26:
try ./configure --enable-non-distributable=yes

SebKuzminsky, 2023-07-01:
Another option is to install `libeditreadline-dev`, which is redistributable under GPL-2.  This is what we do on Debian.

closed 2023-07-01 by SebKuzminsky

nomi-ramzan, 2023-07-02:
@rene-dev After `--enable-non-distributable=yes` I can not find linuxcnc build file.