LinuxCNC/linuxcnc #2559: Loaded G-code program is executed on touch off in some kind of sandbox mode

petterreinholdtsen, 2023-07-02:
I discovered a surprising feature of LinuxCNC I suspect is a bug.  At least it break the principle of least surprise for me.

In short, the loaded G-code program is executed in some kind of sand box mode when pressing the touch off button in the GUI.  I've tested this in Axis.  To reproduce, save this simple g-code program as test-touch-off.ngc:

```
(msg, Did you expect this?)
(print, The value of parameter 1000 is #1000)
(msg, The value of parameter 1000 is #1000)
G53 G0 Z0
G53 G0 X0 Y0
M2
```

Next, load it into your LinuxCNC instance (for example a simulator of axis_mm, but it work on a physical instance too).

Press the 'Home All' button, and then press 'Z' followed by 'Touch Off'.  Notice how the print and msg strings show up in stdout where linuxcnc was started and as notifications in the GUI.

The program was obviously executed at touch off, but the G0 instruction to move is ignored.  I also discovered that the numbered parameter have different content while running in the 'touch off sandbox', and M66 return bogus pin values.

What is going on here?

rene-dev, 2023-07-02:
This is most likely related to the 2 interpreter instances. One is in task, sending commands to motion, the other is running the preview.

andypugh, 2023-07-02:
This is how the system knows to warn if the job does not fit inside the machine limits. And this needs to be recalculated after a touch-off. 
What is a surprise is that the messages get printed during the code check.

petterreinholdtsen, 2023-07-02:
It would be less surprising if the "hidden run" was completely silent, or at least using the real numeric parameter values.  It really threw me off that the numbers I was looking at suddenly were missing...

Btw, notice also how the (msg, The value of parameter 1000 is #1000) line do not print 0 for #1000, it prints '#1000' in the notification.  Is this a bug?  If so, I can create another issue for it.

andypugh, 2023-07-03:
I think it is expected that (MSG, prints verbatim text and (DEBUG, expands parameters. (PRINT, is like (DEBUG, but outputs to stderr. 
http://linuxcnc.org/docs/2.9/html/gcode/overview.html#gcode:messages

PR #2575 (2023-07-10): Get rid of stdout lines from GUI drawing interpreter.
Every time the GUI parse the G code program to draw the expected moves, any (print, message) line will output its message to stdout.  This is very confusing when debugging a program.  This patch rewrite the handling of (print, ...) to avoid this output.

Fixes #2559.
commit 983d68f9 (2023-07-10): Get rid of stdout lines from GUI drawing interpreter.

Every time the GUI parse the G code program to draw the expected moves,
any (print, message) line will output its message to stdout.  This is
very confusing when debugging a program.  This patch rewrite the handling
of (print, ...) to avoid this output.

Fixes #2559.
closed 2023-07-10 by andypugh