LinuxCNC/linuxcnc #2561: GMoccapy user extension ability

c-morley, 2023-07-02:
Quite a few times users have had problems or requests that could be best handled by extending the code of GMoccapy. AXIS, gscreen and qtvcp all have some mechanism of python code extension.

What would be the best/easiest way?
Is there a reason not to do this?

hansu, 2023-07-02:
Gmoccapy already has an option to run user specific python code after GUI creation: http://linuxcnc.org/docs/2.9/html/gui/gmoccapy.html#_user_command_file (ignore the copy and paste typos).
I also think about adding the possibility to run custom code periodically like the other GUIs do.

c-morley, 2023-07-03:
I didn;t see that, thanks.
Is it possible to access the periodic loop somehow? Similar to AXIS: http://linuxcnc.org/docs/2.9/html/gui/axis.html#_user_live_update

hansu, 2023-07-03:
>Is it possible to access the periodic loop somehow? Similar to AXIS: http://linuxcnc.org/docs/2.9/html/gui/axis.html#_user_live_update

Should be possible, but not implemented yet.

zz912, 2023-07-04:
> > Is it possible to access the periodic loop somehow? Similar to AXIS: http://linuxcnc.org/docs/2.9/html/gui/axis.html#_user_live_update
> 
> Should be possible, but not implemented yet.

Hello Hans,

I wanted to ask how difficult it is to implement it?

If it wasn't too difficult, could I ask you for it?

I think Chris wants to help me. (Not only me) I don't see the connections now, but I think it's going to help me solve the problems I have now.

Zdenek

hansu, 2023-07-04:
Yeah it's more or less on my Gmoccapy features-to-add list ;-)

gmoccapy, 2023-10-08:
I do not see the reason to access the periodic part of gmoccapy, as you are able to connect directly to hal status and get the information from there. Add something like a remoteshell or similar might be a good feature, but I would guess one of 1000 unser might use that one.

Imho we should not invest to much time in this issue, as there is some more important code to be updated.
I assigned this issue to HansU and let it open to him to close this issue.

gmoccapy, 2026-10-28:
@hansu close this or not?
IMHO we should close this one

grandixximo, 2026-10-28:
Apologies, confused with another issue, don't mind me

hansu, 2026-10-29:
@gmoccapy Oh, it seems I missed your last comment from 2023. 
I agree we can close this. I don't really see a use case for this so I think we should not add more complexity here just in case.

gmoccapy, 2026-10-29:
Closed, as we do not want to add more complexity

Norb ert

closed 2026-10-29 by gmoccapy