LinuxCNC/linuxcnc #2564: Python Interface command.spindle() does not work

Stutchbury, 2023-07-03:
## Here are the steps I follow to reproduce the issue:

 1. Install Debian 12 and LinuxCNC 2.9 as documented here: 
     http://linuxcnc.org/docs/2.9/html/getting-started/getting-linuxcnc.html#sec:installing_linuxcnc
 2. Open LinuxCNC and create the Gmoccapy SIM (Sample Configurations->sim->gmoccapy->gmoccapy)
 3. Start the Gmoccapy SIM
 4. Run the attached Python file (rename to test-spindle.py) from a terminal.
 As per: 
http://linuxcnc.org/docs/2.9/html/config/python-interface.html#_code_linuxcnc_command_code_methods
(note: there are typos in the spindle docs but I'll raise a separate ticket for that)

## This is what I expected to happen:
Spindle will start at 1024 RPM

## This is what happened instead:
Gmoccapy *SIM* briefly flashes '1024' in the 'Spindle [rpm]' panel and then starts at the default speed set by Gmoccapy.
Gmoccapy prod for my mill (vanilla from Pncconf) does not flash the value but also starts at  a default (currently 60, not sure where from!)

In 2.8 for the Manualmatic pendant I had to implement the 'kludge' (comment out the two lines in the code above) and this would force LinuxCNC to run the spindle at the right speed. This appears to work in the Gmoccapy SIM, but not thing the Pncconf generated config.

I don't think this is a gmoccapy issue as it also fails in a vanilla 
qtdragon config created by Pncconf.

## It worked properly before this:
It worked with the kludge in 2.8.

## Information about my hardware and software:

 * I am using this Linux distribution: Debian GNU/Linux 12 (bookworm):
 * I am using this kernel version: Linux nucnc 6.1.0-9-rt-amd64 #1 SMP PREEMPT_RT Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux
 * I am running: A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
 * I am using this LinuxCNC version: 2.9.0~pre1+git20230208.f1270d6ed7-1
 * I am using this user interface: gmoccapy, 3.4.1
 * I am using this interface hardware vendor and chipset): Mesa Card 7i96 

[test-spindle.txt](https://github.com/LinuxCNC/linuxcnc/files/11940145/test-spindle.txt)

c-morley, 2023-07-03:
Could you give more detail on 'the kludge'

Stutchbury, 2023-07-03:
Hi Chris - yes, sure. In the attached file above there are two lines:
`time.sleep(0.2)`
and then a repeat of the c.spindle() command:
`c.spindle(linuxcnc.SPINDLE_FORWARD, 1024)`

Uncomment them and there's the kludge in all its glory!
I've recently implemented/improved  the spindle handling code in the Manualmatic and was happy with the kludge until I could see if it was fixed in 2.9. Unfortunately it's not only not fixed but the kludge doesn't work either! :-)

c-morley, 2023-07-03:
In the latest 2.9 code, I see this behaivor with GMoccapy but not QtDragon (just using sims not pncconf)
Could you try the builtin sims (I used 'gmoccapy', 'qtdragon_metric', and 'qtdragon_inch') ?

c-morley, 2023-07-03:
If I short cut gmoccapy's spindle speed setting code, Your python program works fine.
It looks like gmoccapy senses the spindle buttons changes regardless if you push them or if it 'pushes' them internally. The python program cause gmoccapy to change the button state and then the the button state code decides to start the spindle at gmoccapy's default speed.
 
If you can get qtdragon to do the same please give as much detail about conditions/states on how to replicate it on my computer.

Stutchbury, 2023-07-03:
Thanks Chris - I've just created the SIM for qtdragon and the spindle behaves as expected. I'll now re-create a qtdragon Pncconfig for my mill (sadly my original NUC bricked itself white doing a BIOS save) and test it - that (like my Pncconf for Gmoccapy+kludge didn't work for the non-SIM config).

c-morley, 2023-07-03:
Ah sorry I didn't read the comments in your python program.
The kluge seems to confirm my idea. Not sure why it doesn't work now as it works in my tests here.

c-morley, 2023-07-03:
maybe you need to increase the kludge sleep time a bit.

Stutchbury, 2023-07-03:
I've just quickly re-re-created my mill's qtdragon config via Pncconf (using the original .pncconf file from 2.8) and that does not work even with the kludge enabled. It seems the SIM's spindle_sim.hal is behaving differently to the .hal generated by Pncconf.
I can upload the .pncconf, .hal * .ini files for my mill if that would help.

Stutchbury, 2023-07-03:
> maybe you need to increase the kludge sleep time a bit.
I had this reliably working at 0.12 in 2.8 but will try some larger values.

c-morley, 2023-07-03:
yes please upload the config files.

Stutchbury, 2023-07-03:
These are the Gmoccapy config files generated by Pncconf (from the original 2.8 .pncconf) - qtdragon to follow.

[wm18-gmoccapy-v1.hal.txt](https://github.com/LinuxCNC/linuxcnc/files/11940621/wm18-gmoccapy-v1.hal.txt)
[wm18-gmoccapy-v1.ini.txt](https://github.com/LinuxCNC/linuxcnc/files/11940622/wm18-gmoccapy-v1.ini.txt)
[wm18-gmoccapy-v1.pref.txt](https://github.com/LinuxCNC/linuxcnc/files/11940623/wm18-gmoccapy-v1.pref.txt)

Stutchbury, 2023-07-03:
And here's the whole qtdragon folder - I've commented out the Manualmatic in custom.hal so should load as vanilla from Pncconf.

[wm18-qtdragon-v1.zip](https://github.com/LinuxCNC/linuxcnc/files/11940651/wm18-qtdragon-v1.zip)

Stutchbury, 2023-07-04:
A little more investigation this morning and it would appear neither my Pncconf generated qtdragon or gmoccapy config control the spindle from the GUI either - so looks like that may be a Pncconf issue.
I'm very conscious this is not a support area so will move that issue to the forum - although I think you've already found a 'real' issue with the gmoccapy spindle button sensing, so I don't feel too guilty :-) 
As an aside, the gmoccapy 'X-' jog button does not work in continuous mode on my config (X+, Y & Z are all fine) - it'll do a small step and then stop.

rmu75, 2025-12-10:
Is this stiff relevant?

rmu75, 2026-01-07:
fixed

closed 2026-01-07 by rmu75