LinuxCNC/linuxcnc #2572: linuxcnc main command is missing after successfully built

nomi-ramzan, 2023-07-09:
## Here are the steps I follow to reproduce the issue:

 1. Install Fedora IOT 38 and following [this link](https://linuxcnc.org/docs/html/code/building-linuxcnc.html)
 2. I Installed dependencies `git-core autoconf automake gcc gcc-c++ tcl-devel python-Yapps2 python3-devel python3-tkinter python3-argcomplete libtirpc-devel libudev-devel libmodbus-devel libusb1-devel glib2-devel gtk3-devel intltool boost-devel readline-devel mesa-libGLU-devel libXmu-devel asciidoc tk-devel bwidget`
 3. git clone https://github.com/LinuxCNC/linuxcnc.git linuxcnc-dev
 4. cd linuxcnc-dev/src/
 5. autoreconf --install
 6. ./autogen.sh
 7. ./configure --with-realtime=uspace --enable-non-distributable=yes

## This is what I expected to happen:
Successfully build linuxcnc for Fedora
## This is what happened instead:
Missing /bin/linuxcnc command. Rest list in bin directory is
```
5axisgui       gremlin               halrmt       hy_vfd
axis           gremlin_view          halsampler   image-to-gcode
axis-remote    gs2_vfd               halscope     inivar
classicladder  gscreen               halshow      io
debuglevel     hal_input             halstreamer  iov2
elbpcom        hal_manualtoolchange  halui        lineardelta
genserkins     halcmd                hbmgui       lintini
gladevcp       halcompile            hexagui      linuxcnc_module_helper  
gmoccapy       halmeter              hy_gt_vfd    linuxcnclcd
linuxcncrsh    milltask              pmx485       qtplasmac-cfg2prefs
linuxcncsvr    millturngui           pmx485-test  qtplasmac-materials
linuxcnctop    mitsub_vfd            pncconf      qtplasmac-plasmac2qt
maho600gui     modcompile            profile_axis qtplasmac_gcode
max5gui        motion-logger         puma560gui   qtvcp
mb2hal         mqtt-publisher        pumagui      rotarydelta
mdi            ngcgui                pyngcgui     rs274
mdro           panelui               pyui         rtapi_app
melfagui       pi500_vfd             pyvcp        scaragui
schedrmt       tool_mmap_read        wj200_vfd
scorbot-er-3   tool_watch            xhc-hb04
sendkeys       tooledit              xhc-whb04b-6
shuttle        touchy                xyzab-tdr-gui
sim-torch      tracking-test         xyzac-trt-gui
stepconf       update_ini            xyzbc-trt-gui
svd-ps_vfd     userspace_count_names z_level_compensation
teach-in       vfdb_vfd
thermistor     vfs11_vfd
```

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `cat /etc/`): 
   ```
    NAME="Fedora Linux"
    VERSION="38.20230619.0 (IoT Edition)"
    ID=fedora
    VERSION_ID=38
    VERSION_CODENAME=""
    PLATFORM_ID="platform:f38"
    PRETTY_NAME="Fedora Linux 38.20230619.0 (IoT Edition)"
    ANSI_COLOR="0;38;2;60;110;180"
    LOGO=fedora-logo-icon
    CPE_NAME="cpe:/o:fedoraproject:fedora:38"
    HOME_URL="https://fedoraproject.org/"
    DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f38/system-administrators-guide/"
    SUPPORT_URL="https://ask.fedoraproject.org/"
    BUG_REPORT_URL="https://bugzilla.redhat.com/"
    REDHAT_BUGZILLA_PRODUCT="Fedora"
    REDHAT_BUGZILLA_PRODUCT_VERSION=38
    REDHAT_SUPPORT_PRODUCT="Fedora"
    REDHAT_SUPPORT_PRODUCT_VERSION=38
    SUPPORT_END=2024-05-14
    VARIANT="IoT Edition"
    VARIANT_ID=iot
    OSTREE_VERSION='38.20230619.0'
```
 * I am using this kernel version (shown by `uname -a`): Linux localhost.localdomain 6.3.8-200.fc38.aarch64 SMP PREEMPT_DYNAMIC Thu Jun 15 02:15:18 UTC 2023 aarch64 GNU/Linux
 * I am running ...
   - [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   - [x] A binary I built myself
   - [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.10.0-pre0-1542-g93fccc095b
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): GPIO on raspberry pi 4b 8GB

andypugh, 2023-07-09:
"linuxcnc" is a script in the /scripts directory. 

Are you saying that the script isn't found even after rip-environment?

rene-dev, 2023-07-09:
its generated by autoconf from this file: https://github.com/LinuxCNC/linuxcnc/blob/master/scripts/linuxcnc.in
looks like there are more files missing.

rodw-au, 2023-07-09:
Did you actually type `make` between 7 and 8 ?
the list of dependencies appears short. DId you get them using `dpkg-checkbuilddeps` ?

andypugh, 2023-07-09:
Have you looked in the "scripts" directory? 

This issue is going to be very hard to close if you don't reply.

closed 2023-07-10 by nomi-ramzan