LinuxCNC/linuxcnc #2586: Python command.wait_complete() does not wait for c++ halui commands

zz912, 2023-07-16:
I use RIP installation branche 2.9. 
I use Gmoccapy only for simulate problem. This is NOT problem of Gmoccapy.

For simulation bug I use the coolant button in Gmoccapy. I added this code after this line:
https://github.com/LinuxCNC/linuxcnc/blob/c4e94293683fc459124f7b25adde26ddd85c1086/src/emc/usr_intf/gmoccapy/gmoccapy.py#L4127
```
            self.command.wait_complete(30)
            self.command.mode(linuxcnc.MODE_MDI)
            self.command.wait_complete(30)
            self.command.mdi("G91 G1 X10 F200")
            self.command.wait_complete(30)
            self.command.mdi("G91 G1 Y10 F200")
            self.command.wait_complete(30)           
            self.command.mode(linuxcnc.MODE_MANUAL)
            self.command.wait_complete()
```
I added the following to the Gmoccapy.ini file:
```
[HALUI]
MDI_COMMAND = G91 G1 Z-10 F200
```
[image]
In the video you can see that command.wait_complete() does not wait for halui to complete its commands. It will run python commands and thus LCNC will fall into an error.

I think that this bug was caused by the fact that, when creating Python Interfaces, they forgot that there are halui MDI commands:
http://linuxcnc.org/docs/2.9/html/gui/halui.html#_mdi_2

More information here:
https://forum.linuxcnc.org/38-general-linuxcnc-questions/49445-python-command-wait-complete-does-not-wait-for-c-halui-commands

andypugh, 2023-07-27:
Sorry, as the video loops it isn't easy to work out what the sequence of events is. Can you describe that sequence?

zz912, 2023-07-28:
I know the loop is not good. When I used screen recording to mp4, github had a problem with it. The same videos were displayed correctly on some PCs and on some PCs it reported that they were corrupted.

Sequence starts when I set HAL pin in HALSHOW.
Sequence stops when message "Must be in ..." is shown.

It is also not clear from the video that as soon as I start the halui MDI-command, I immediately click on the "Cooling" button. The "Cooling" button turns green only after a certain time after clicking.

There is no sequence of normal use of the LCNC in the video. It's a simulation of a bug.
I intentionally create a "race condition" in the video.

The important information is that even Chriss Morley thinks this is a problem. He wrote it on the forum.
That's why I put it here so that this problem is documented.

gmoccapy, 2023-07-28:
IMHO we need to search for the error /race condition etc as follows:
Setting the Hal pin, sends the MDI command. Gmovvapy reacts correctly, as it changes to MDI mode. 
Now you are able to see, that the macro does not stop (button in the bottom line most right has still the stop icon)
Now he clicks on the Cooling button, witch is modified as far as I remember to also send a MDI command.

Gmoccapy stays in MDI screen design and LinuxCNC has changed back to manual mode, but gmoccapy did not get any notice / Information of the mode change.

That is IMHO the reason of the annoying message.

As mentioned, that is a case study in September at the LinuxCNC meeting

Norbert

zz912, 2023-12-30:
> As mentioned, that is a case study in September at the LinuxCNC meeting

Hello Norbert,

I wanted to ask if this problem was solved at the meeting?

I'm still having problems integrating Gmoccapy + LCNC on the machine. The problems mainly concern the ATC control functions. These functions behave randomly. I'm worried that they might forget to turn on corrections (G43). I don't feel good about this Gmoccapy + LCNC integration.

The goal of this Issue is not to prove the imperfection of Gmoccapy + LCNC. But it's the easiest way to simulate the source of my problems. I think this Issue is the source of my problems.

I am constantly fighting with remap + [HALUI] MDI_COMMANDS + Gmoccapy + my poor programming skills.

I don't know how to proceed at the moment. I'm surprised I'm the only one struggling with [HALUI] MDI_COMMANDS. I think I don't use anything that other people don't need too.

Zdenek