LinuxCNC/linuxcnc #2587: LinuxCNC 2.9 MDI move commands bugged and zeroing after MDI commands bugged.

bhowiebkr, 2023-07-16:
### Not sure if this is 2 bugs or one in the same:

1. When issuing move MDI commands on a single axis it'll move other axis at the same time. 
2. Also when zeroing out after an MDI move command that axis will not correctly zero out needing me to press the button twice. 

**See this screen recording of my machine running:**
[image]

## This is what I expected to happen:
1. When issuing an MDI command like `g0x10` the machine should only move on the X-axis and not on other axes.
2. When zeroing out a g54 coord axis, I should not have to press it twice. 

## This is what happened instead:
see video above. 
1. An MDI move on X will move the Y axis as well. Same when moving just the Y axis.
2. Zeroing out an axis after an MDI move in g54 coords requires me to press the button twice as a single press will give me a garbage value.

## It worked properly before this:
On a difference machine with Linuxcnc 2.9 built from source from last year non of this happened.

## Information about my hardware and software:

### Dist:
```shell
howard@debian12:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 12 (bookworm)
Release:	12
Codename:	bookworm
```

### kernel
```shell
howard@debian12:~$ uname -a
Linux debian12 6.1.0-10-rt-amd64 #1 SMP PREEMPT_RT Debian 6.1.37-1 (2023-07-03) x86_64 GNU/Linux
```

- This is on the Debian released package
- Happening in qt dragon hd interface, but when talking to other people on the PrintNC discord, some have had it happen in probe basic as well.
- I'm using a mesa 7i76e to interface with the machine.

bhowiebkr, 2023-07-16:
here are my machine's configs:
https://github.com/bhowiebkr/PrintNC_conf/tree/main/PrintNC

andypugh, 2023-07-16:
That is a bit wierd...

Is the UI set to display actual or commanded position? 

(Not that there should really be any difference with a stepper machine) 

But it will help to track down the problem if we know if this is on the command or the feedback side.

bhowiebkr, 2023-07-16:
I just tested this in my virtual machine of debian 12 running a default simulator profile and linuxcnc 2.9 debian build. Exact same thing happens there as well on the simulator.

bhowiebkr, 2023-07-16:
> That is a bit wierd...
> 
> Is the UI set to display actual or commanded position?
> 
> (Not that there should really be any difference with a stepper machine)
> 
> But it will help to track down the problem if we know if this is on the command or the feedback side.

This would be command position I guess? I'm running servos in step position and have no feedback of position to linuxcnc. Also in the simulator with a default dragon interface (not my machine configs) the same bugs show up. I haven't tried it with other interfaces like probebasic in the simulator.

bhowiebkr, 2023-07-16:
I checked in the simulator gmocappy and AXIS and they both appear to work correct.

andypugh, 2023-07-17:
Do the same commands have the same, or the correct, effect if run from a G-code programme?

Taking the example from the video of G0 X10 / G0 Y7 leaving the axes at X-0.068 Y=7, I assume that replacing the second G0 with G0 X10 Y7 works properly? 

(I am currently compiling a snapshot of the code from f1270d6 (which rather usefully appears in the splash screen)

I don't currently see anything unexpected running the latest 2.9 in Debian Buster. (OS really shouldn't be important)

andypugh, 2023-07-17:
I can reproduce the problem. 
I giess some good news is that it seems that it got fixed somewhere between the snapshot in Debian and the current version of 2.9. 
I will investigate further tomorrow, it's late here now. 
As a workaround for your particular system you could try installing the latest version from the buildbot
http://buildbot.linuxcnc.org 
But it's probably worth waiting for me to find out exactly when the problem got fixed. (If, indeed, it did)

TheRealBeef, 2023-07-17:
I'm just writing to confirm that I was able to reproduce the bug of not properly zeroing coordinates using the Probe Basic GUI, but I don't believe it's a GUI dependent issue. On LinuxCNC 2.9 of course. It's not just a display issue as it resulted in having incorrect coordinates when moving to the next location on my machine.

The workaround for me was to ensure I zeroed the coordinates twice, which I've been doing pretty religiously since identifying this, and I hadn't thought to look beyond that.

If it's been fixed since 2.9 then it's quite nice, but it's definitely worth the issue report here.

andypugh, 2023-07-17:
> I'm just writing to confirm that I was able to reproduce the bug of not properly zeroing coordinates using the Probe Basic

Probe Basic is run as a separate project, it might be worth raising that issue there. 
https://github.com/kcjengr/probe_basic/issues

TheRealBeef, 2023-07-17:
Thanks, but as mentioned above and in the original issue I don't believe it's a GUI specific issue since it occurs in other GUIs that aren't a separate project. It most likely is/ (or rather was, since as you mentioned it is apparently already resolved in LinuxCNC snapshot build) a LinuxCNC issue.

bhowiebkr, 2023-07-17:
> I can reproduce the problem. I giess some good news is that it seems that it got fixed somewhere between the snapshot in Debian and the current version of 2.9. I will investigate further tomorrow, [...]

I’ll hold off on buildbot builds and wait to see what you come up with. On my old PC/linuxcnc controller box I built it from source from the main branch on github. Never ran into any problems. I’ll get that machine booted back up tonight so I can check what was the last source commit that I pulled. Was some time last year. Might help track down what had changed between that commit and now. 

Maybe something specific with QtVCP as dragon and probe basic both have dependencies of it. gmocappy and AXIS run correct in simulator mode. I tried a bunch of others but many have breaking issues on bootup (some of the new ones have missing python dependencies in lcnc2.9)

andypugh, 2023-07-17:
OK, after setting up a new Bookworm VM as I couldn't persuade 2.9 to build on Buster, I did a git-bisect and tracked the fix down to this commit:
https://github.com/LinuxCNC/linuxcnc/commit/aaffb434167fd9600b26c65bb339f94738bffa40
Which was introduced as a fix for 
https://forum.linuxcnc.org/qtvcp/48337-qtdragon-mdi-jogging-strange-diagonals

We will work on getting this fix into Debian ASAP.

In the meantime the behaviour can be corrected on your machine by saving this file:
https://raw.githubusercontent.com/LinuxCNC/linuxcnc/aaffb434167fd9600b26c65bb339f94738bffa40/lib/python/qtvcp/widgets/mdi_line.py

Into this location on your HD: /usr/lib/python3/dist-packages/qtvcp/widgets/mdi_line.py 

ie
````
sudo mv mdi_line.py /usr/lib/python3/dist-packages/qtvcp/widgets/mdi_line.py 
````

bhowiebkr, 2023-07-17:
I’ve checked and verified that this fixes it for qt dragon. Both bugs were fixed.  @TheRealBeef  This will very likely fix it for probe basic as it's using the mdi widget from qtvcp as a ui element. 

@andypugh Thanks for your help! I'll let you close this issue if you want to have it tracked till it's pushed out to places like Debian packages.