LinuxCNC/linuxcnc #2589: Crash to desktop on hitting cycle start

mdkendall, 2023-07-18:
LinuxCNC installed on HP Prodesk 800 G2 from LinuxCNC 2.8.4 Debian 10 Buster PREEMPT-RT ISO, controlling a 3040Z-DQ mill via a parallel port card. Installed and configured in May, has been working fine since then.

This morning I opened a .nc file as usual and hit the cycle start button, LinuxCNC closed and dropped me to the desktop, and a crash report was produced. On restarting the program (and re-homing the machine) everything works again without issue.

## Here are the steps I follow to reproduce the issue:

- Issue is not reproducible. Happened once on hitting cycle start.

## This is what I expected to happen:

- Job runs.

## This is what happened instead:

- LinuxCNC closed
- Crash report produced

## It worked properly before this:

- No recent changes

## Information about my hardware and software:

 * I am using this Linux distribution and version : `Debian Debian GNU/Linux 10 (buster) 10 buster`
 * I am using this kernel version : `#5 SMP PREEMPT Sun Jul 11 19:13:27 BST 2021`
 * I am running ...
   * [X] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version : `2.8.4`
 * I am using this user interface (GUI) : `AXIS`
 * I am using this interface hardware vendor and chipset : "StarTech.com 1-Port Parallel PCIe Card - PCI Express to Parallel DB25 Adapter Card - Desktop Expansion LPT Controller for Printers, Scanners & Plotters - SPP/ECP - Standard/Low Profile". Model number PEX1P2. Based on ASIX AX99100 chip.

Crash report is attached.
[linuxcnc.report.txt](https://github.com/LinuxCNC/linuxcnc/files/12087440/linuxcnc.report.txt)

andypugh, 2023-07-18:
There is nothing in the report that I can interpret as saying what went wrong. 

We see: "can't do that (EMC_TRAJ_SET_TELEOP_ENABLE) in auto mode with the interpreter waiting" (which is a little mysterious, and has been seen before, but the cause hasn't been determined as far as I recall) 

And also
"X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  15 (X_QueryTree)
  Resource id in failed request:  0x2d216d8
  Serial number of failed request:  6297929639
  Current serial number in output stream:  6297929639"

Which normally indicates a problem with the windowing system. 

Perhaps you could start LinuxCNC from the command line (command "linuxcnc") and compare the output there to what is seen during the crash dump. Anything that is different is suspicious.

mdkendall, 2023-07-18:
Thanks for looking at this.

The machine had been up with LinuxCNC running for about two months before the crash, so I'm not sure if any errors in that log were associated with today's issue, or were historic. Since the issue is not reproducible (it has run a few jobs today since the crash with no problems) I'm not sure I can do much to debug.

Feel free to close this issue. I will keep an eye on it and re-open if it happens again.