LinuxCNC/linuxcnc #2593: tooledit Rounding issue on tool offset

robharp, 2023-07-21:
## Here are the steps I follow to reproduce the issue:

 1. open tooledit to edit a tool length
 2. make a tool 3 digital number proceeded with more than 3 decimals  123.1596
 3. the number is rounded up when using a 3 digit proceeding number to 3 decimal places, 
    -  but with less than 3 proceeding length number its rounded to deciaml 8 places
 

## This is what I expected to happen:
all numbers should be rounded to 8 Decimal or better regarding what the proceeding number is.  
*note have not comfirmed this on an imperial setup only metric
## This is what happened instead:

see forum post for more info, also confirmed by andy
https://forum.linuxcnc.org/38-general-linuxcnc-questions/49558-tooledit-rounding-issue-on-tool-offset

## It worked properly before this:
unknown assuming its always been like this, but users with long tool lengths would only see this issue..
## Information about my hardware and software:

Linuxcnc 2.8.4
Metric Mode set in ini
Diamiter programing on
Gui : Axis
mode : Lathe

d2inventory, 2023-07-28:
The issue is the %g formatting used.

This has other "issues", f.e. 0.000054 would be written as 5.6e-05

Using %.9g would limit the number of digits to 9 (instead of 6 as is the default for %g) and fix this issue,
but the other issue I mentioned would still be there.

Code line: https://github.com/LinuxCNC/linuxcnc/blob/ab8778c16e4ff069e41ef647b3326e0c109191b3/tcl/tooledit.tcl#L99

grandixximo, 2026-10-05:
Fixed in 5d6e5eac0a (2023-09-08, in 2.9.0): the tool table format changed from %g to %.9g as suggested above, with a regression test in tests/tooledit. Closing.

closed 2026-10-05 by grandixximo