LinuxCNC/linuxcnc #2598: PNCConf incorrect HAL out for inverted SSR and OUTM pins

dpslwk, 2023-07-25:
PNCConf build_HAL is not correctly inverting pins for SSR / OUTM pins

Result from PNCConf
```
setp [HMOT](CARD0).outm.00.out-05.invert_output true
setp [HMOT](CARD0).ssr.00.out-00.invert_output true
```

I believe this should be 
```
setp [HMOT](CARD0).outm.00.invert-05 ture
setp [HMOT](CARD0).ssr.00.invert-00 true
```

## Here are the steps I follow to reproduce the issue:

 1. PNCConf test coinfig 
 2. Mesa Card 0-> Board name: `7i96s-Internal Data`, Firmware: `7i96s_d` 
 3. TB3: define a function for 01 `SSR Output` (`Control->Machine Is Enabled`)  and click `Inv`
 4. TB3: define a function for 04 `OUTM Output` (`Spindle->Spindle Enable`) and click `Inv` 
 5. finish and review build Hal file
 
## This is what I expected to happen:
```
# --- SPINDLE-ENABLE ---
net spindle-enable  =>     [HMOT](CARD0).outm.00.out-04
setp [HMOT](CARD0).outm.00.invert-04 ture
```

## This is what happened instead:
```
# --- SPINDLE-ENABLE ---
net spindle-enable  =>     [HMOT](CARD0).outm.00.out-04
setp [HMOT](CARD0).outm.00.out-05.invert_output true
```

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`): Debian GNU/Linux 12 (bookworm)
 * I am using this kernel version (shown by `uname -a`): Linux 6.1.0-10-rt-amd64 #1 SMP PREEMPT_RT Debian 6.1.37-1 (2023-07-03) x86_64 GNU/Linux
 * I am running ...
   * [x] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): linuxcnc-uspace_2.9.0~pre1+git20230208.f1270d6ed7-1_amd64.deb
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): gmoccapy
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card): MESA 7i96S

c-morley, 2023-07-28:
Well that's horrible - someone broken the naming scheme for SSR and outM .
Can you post what a normal not inverted pin name is for these pins?

dpslwk, 2023-07-28:
not sure what you mean by not inverted pin names?

two commits that added inverting to SSR and the full OUTM module
https://github.com/LinuxCNC/linuxcnc/commit/dbaa51e484453541fdcfbff5ddbafeae445cb3cd
https://github.com/LinuxCNC/linuxcnc/commit/9f3109f242002723b15ab95481dbcdd59091e097

[image]
[image]

Since I suspect its far two late to rename these in hostmot2 I think a fix would be needed in `build_HAL.py`
https://github.com/LinuxCNC/linuxcnc/blob/db02b5b27f35cbde6cd73c2ec2da0e7d431528ec/src/emc/usr_intf/pncconf/build_HAL.py#L1689-L1694

c-morley, 2023-07-28:
Ya so I needed to see the normal pin name to see how the scheme works.
every other pin AFAIK adds some sort of invert text after the normal pin name, these but the invert text before the number. So the fix needs to go deeper in the code.
Thanks for the print out that should give me what I need.

dpslwk, 2023-07-28:
I'm not setup to build dev and test actual changes myself, but I suspect this might work? 

``` python
if i: # invert pin
    if t in (_PD.SSR0,_PD.OUTM0):
        print("setp %s true"%  (pinname.replace('out-', 'invert-')), file=file)
    else:
        if "sserial" in pname:
            ending = "-invert"
        elif "parport" in pinname: ending = "-invert"
        else: ending = ".invert_output"
        print("setp %s true"%  (pinname + ending ), file=file)
 ```

c-morley, 2023-07-28:
Nice ! I'll test it out.

c-morley, 2023-07-28:
Ok that looks good - how about inverted inputs - are they ok?

dpslwk, 2023-07-28:
INM is fine, that’s seems to use more standard `-not`

c-morley, 2023-07-28:
thank you. I will push your fix to 2.9. I'll close this. please open it again if this somehow didn't work

closed 2023-07-28 by c-morley