LinuxCNC/linuxcnc #2607: Random toolchanger -> multiple tools at same pocket

forrestg, 2023-08-07:
Hello,

about half a year about I report that is not possible to have multiple tools at same pocket if random tool changer is set.
I meat that andypugh told me thats becouse of indexes at tool table.

I'm not going to put up with it so I rewrite some code change logic for storing and reading tools from tool table and
now its working. Tested at gmoccapy and 2.9.0~pre1.

Is it chance to add it into lcnc ?

Thanks Michal

rene-dev, 2023-08-07:
Im probably the one who knows most about the problems with the tooltable, and I also have a pr which isn't quite ready. Maybe your code helps, do you have it in a branch somewhere?

forrestg, 2023-08-07:
My code changing tools good but I must fish changing tool comments too. After I have it I can push code to place what you want. Tell me what its best place for you?

forrestg, 2023-08-12:
I add repository with changes https://github.com/forrestg/linuxcnc-tooltable