LinuxCNC/linuxcnc #2613: Gmoccapy - offset Z - wrong after tool 0

zz912, 2023-08-23:
RIP 2.9 Branche
Debian Bookworm
Oracle VM VirtualBox
Test sim Gmoccapy.

If any tool other than 0 is selected at the start, the z offset is displayed correctly.
If I choose tool 0, the z offset is displayed incorrectly.
[image]

rmu75, 2023-08-23:
Before selecting tool 0, G43 is active, after you select tool 0, G49 is active, i.e. tool compensation is off.

zz912, 2023-08-23:
So the question is why is G43 automatically disabled when tool 0 is selected?

zz912, 2023-08-30:
I did some research on this issue and found out that it is again a problem with the G43 auto-starting after a tool change.

https://github.com/LinuxCNC/linuxcnc/blob/c2b55957e91242ccddfb70c68bf0b59b783c19cd/src/emc/usr_intf/gmoccapy/gmoccapy.py#L3540-L3544

Norbert knows about this problem and plans to solve it. However, I will leave this Issue open so that we don't forget to check if his solution removes this consequence as well.

My solution is:
1) Remove lines auto-starting g43
2) Add the lines here:
https://github.com/LinuxCNC/linuxcnc/blob/c2b55957e91242ccddfb70c68bf0b59b783c19cd/src/emc/usr_intf/gmoccapy/gmoccapy.py#L4980-L4982
```
            self.command.wait_complete()
            self.command.mdi("G43")
            self.command.wait_complete()
```

rmu75, 2023-08-30:
I don't see how "tool offset == 0" can influence the code you cited.

zz912, 2023-08-30:
The cited code makes random problems. It makes cace conditions. I have been dealing with this for a long time.

lookt at:
https://github.com/LinuxCNC/linuxcnc/pull/2552

lookt at:
https://forum.linuxcnc.org/38-general-linuxcnc-questions/49445-python-command-wait-complete-does-not-wait-for-c-halui-commands

lookt at:
https://github.com/LinuxCNC/linuxcnc/issues/2453#issuecomment-1565086089

I would like ask you. Could you try reproduce this bug?  I wouldn't be surprised if I was the only one with the problem.

rmu75, 2023-08-30:
This issue may be related but I don't think this is caused by a race condition.

zz912, 2023-08-30:
Feel free to find source of problem. I am mechanical designer. Programing is only my hobby.

gmoccapy, 2023-08-31:
In Axis you are changing the tools by MDI commands like T1 M6 and T2M6, but in gmoccapy you use M61 Q1 and M61 Q2.

So that is one difference. I checked gmoccapy code and did not find any place, where M49 is commanded. So the problem may be deeper.

Norbert

zz912, 2023-08-31:
> In Axis you are changing the tools by MDI commands like T1 M6 and T2M6, but in gmoccapy you use M61 Q1 and M61 Q2.

Thank you for notice. Here is AXIS with the M61 QX:
[image]
G49 does not appear