LinuxCNC/linuxcnc #2614: AXIS - RIP installation wants outside file

zz912, 2023-08-23:
RIP 2.9 Branche
Debian Bookworm
Oracle VM VirtualBox
Test sim AXIS

I have RIP installation of LinuxCNC. If I wants "Edit tool data..." It wants file from folder, witch is NOT in RIP folder.
 
[image]

d2inventory, 2023-08-24:
I can't reproduce the error. Which commit are you using?

git rev-parse HEAD

prints the current active commit hash

zz912, 2023-08-24:
I dont use git.
I dowloaded linuxcnc-2.9.zip
[image]

I fixed it for myself:
[image]

d2inventory, 2023-08-24:
Where did you get that zip from?

zz912, 2023-08-24:
[image]

zz912, 2023-08-24:
I dont know, what happened. I cannot reproduce this bug again. I started from clean unzip folder.
Thank you for help.  I apologize for the confusion.

closed 2023-08-24 by zz912

zz912, 2023-08-29:
Yes, I have icon in desktop for terminal with Rip-environment. I started linuxcnc throw this icon.