LinuxCNC/linuxcnc #2616: Motion - wrong protect for tandem axis "HOME_SEQUENCE = -0"

zz912, 2023-08-25:
RIP 2.9 Branche
Debian Bookworm
Oracle VM VirtualBox

Hello,

Maybe I'll make an interesting machine with "X Y1 Y2 Z1 Z2" kinematics.
That's why I'm preparing for this situation. I made the configuration "X Y1 Y2 Z1 Z2" by PncConf. I used it to simulate this configuration. Here are configuration sim files:
[Z0069_sim-003.zip](https://github.com/LinuxCNC/linuxcnc/files/12435799/Z0069_sim-003.zip)

The homing sequence must be:
1) Z
2) X
3) Y

This creates a situation where the INI file contains:
HOME_SEQUENCE = -0
Minus zero is not my invention. This was invented by PncConf.

The bug is that I can break the CNC machine in joint mode. I can move the Z axes asynchronously.
[image]

I think the bug is here:
https://github.com/LinuxCNC/linuxcnc/blob/9788773b9a0a1677ee4f91b15dff3b14829b4a7b/src/emc/motion/homing.c#L219-L221
Condition "ABS(H[jj].home_sequence" ignore  "HOME_SEQUENCE = -0"

andypugh, 2023-08-25:
-0 is valid in IEEE floating point. But .home_sequence is an int, so I agree that the setup wizard should not be using -0. 
(Or, possibly, should refuse to use 0 as the sequence number for a tandem joint. I haven't run a test to see how this happens)

zz912, 2023-08-25:
I looked in the documentation:
http://linuxcnc.org/docs/2.9/html/config/ini-homing.html#sec:homing-section

> The initial HOME_SEQUENCE number may be 0, 1 (or -1).

This is NOT bug of motion, but this is a bug of PncConf.

Correct solution for my exmple is:
[JOINT_0]HOME_SEQUENCE = 2
[JOINT_1]HOME_SEQUENCE = -3
[JOINT_2]HOME_SEQUENCE = -3
[JOINT_3]HOME_SEQUENCE = -1
[JOINT_4]HOME_SEQUENCE = -1

This works correct.

zz912, 2023-10-01:
https://github.com/LinuxCNC/linuxcnc/pull/2660 covers this problem.

closed 2023-10-01 by zz912