LinuxCNC/linuxcnc #2619: motion.motion-type is 0 when rigid tapping

curtdutt, 2023-08-26:
During a rigid tapping operation motion.motion-type remains at 0.

This is because the tcInit call on line 1589 src/emc/tp/tp.o is passing 0 to canon_motion_type.

I propose changing the 0 to EMC_MOTION_TYPE_FEED.

 tcInit(&tc,
            TC_RIGIDTAP,
            0,
            tp->cycleTime,
            enables,
            1);

becomes

 tcInit(&tc,
            TC_RIGIDTAP,
            2,
            tp->cycleTime,
            enables,
            1);

After testing with a live machine there appears to be no negative effects.

petterreinholdtsen, 2023-08-26:
[Curtis Dutton]
> I propose changing the 0 to EMC_MOTION_TYPE_FEED.

Can you explain why this would be useful?

I do not know the code, just curious.
-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2023-08-28:
I think that it would make sense for spindle-synched motion to be a motion-type of its own. But a quick check shows that G33 and G76 moves show as motion type 2. 

This seems like a bug, and should maybe be fixed in 2.8 first.

curtdutt, 2023-09-08:
A motion type of its own would be just fine with me as well and probably
more "correct".

I have a component that depends upon the motion-type going to something
other than 0 when rigid tapping is occurring. It collects cutting time data
on the running job and because of the but outputs no data for my taps as
motion type doesn't change.

c-morley, 2023-10-01:
Could add:
#define EMC_MOTION_TYPE_RIDGIDTAP 7
to nml_intf/motion_types.h

andypugh, 2023-10-02:
> Could add: #define EMC_MOTION_TYPE_RIDGIDTAP 7 to nml_intf/motion_types.h

I feel that all spindle-synched motions should be in the same motion group. 

I think I will change it to 2 in 2,9 to make it consistent with G76 and G33, and then add a new type in Master.

andypugh, 2023-10-02:
Setting G76, 33, 33.1 to a new type is harder than I thought, as they are a STRAIGHT_TRAVERSE with a modifier on the feedrate. 

So, for the moment, I will close this as fixed by 5a14548f303cca6e7f40ada6a43ad5ea1afbe174

closed 2023-10-02 by andypugh