LinuxCNC/linuxcnc #2627: Gmoccapy - box_dro_side is displayed in the G-code editor

zz912, 2023-09-05:
RIP 2.9 Branche
Debian Bookworm
Oracle VM VirtualBox

I added my own box_dro_side in Gmoccapy:
INI file:
EMBED_TAB_NAME = Axis load
EMBED_TAB_LOCATION = box_dro_side
EMBED_TAB_COMMAND = gladevcp -x {XID} -H ./glade/axis_load.hal ./glade/axis_load.glade
[axis_load.zip](https://github.com/LinuxCNC/linuxcnc/files/12520267/axis_load.zip)
[image]

Problem is that  box_dro_side is displayed in the G-code editor too:
[image]

gmoccapy, 2023-09-05:
valid values are:

    ntb_user_tabs (as main tab, covering the complete screen)'

    ntb_preview (as tab on the preview side)'

    box_left (on the left, complete high of the screen)

    box_right (on the right, in between the normal screen and the button list)

    box_coolant_and_spindle (will hide the coolant and spindle frames and introduce your glade file here)

    box_cooling (will hide the cooling frame and introduce your glade file)

    box_spindle (will hide the spindle frame and introduce your glade file)

    box_vel_info (will hide the velocity frames and introduce your glade file)

    box_custom_1 (will introduce your glade file left of vel_frame)

    box_custom_2 (will introduce your glade file left of cooling_frame)

    box_custom_3 (will introduce your glade file left of spindle_frame)

    box_custom_4 (will introduce your glade file right of spindle_frame)

So you made a new place to put the glade panel in, correct?
If so, you did modify the gmoccapy.glade file and there you may have made an error in the order.

How did you modify the gmoccapy.glade file?

Norbert

gmoccapy, 2023-09-05:
And adding a new location also implicit a change in gmoccapy.py code.

see:
def set_up_user_tab_widgets(self, tab_locations):
and
def _init_dynamic_tabs(self):

Norbert

zz912, 2023-09-05:
I use original gmoccapy.glade from branche 2.9 .

It is not documented.

It made probably HansU.

It was my wish:
https://forum.linuxcnc.org/gmoccapy/48951-need-some-opinions-about-the-gmoccapy-gui#270233

hansu, 2023-09-05:
Yes I added this tab location and it's documented (https://linuxcnc.org/docs/2.9/html/gui/gmoccapy.html#gmoccapy:configuration-of-tabs-and-side-panels).
But it seems that I have done it not correctly :/.

PR #2813 (2023-12-31): Gmoccapy 3.4.7 - some improvements
 - Added setting to allow smaller width of virtual keyboard (for widescreens)
[image]

 - Fixed box_dro_side is displayed in the G-code editor (fixes #2627)
 - Fixed error on error message when sound is not available (reported on Raspberry Pi)
 - Fixed increasing of window size when clicking edit button
closed 2024-01-06 by hansu