LinuxCNC/linuxcnc #2629: If abort homing in limit, then machine on the machine will rehoming itself

yuyue2013, 2023-09-06:
## Here are the steps I follow to reproduce the issue:
 1. in ini file i had setting the home all have sequence, and the limit sensor also use for home
 2. home all, and when have axis in limit then i press esc button to abort homing, then the machine will in limit
 3. check on the override limit, then put on machine. the machine will do home again it self...

## This is what I expected to happen:
 I want after homing abort in limit,  if i override limit and machine on, the machine don't do home itself

## It worked properly before this:
This issue present on 2.10

## My solution:
 in homing.c file have the function:  ```static void base_do_cancel_homing(int jno) ```
 i change it to:
```cpp
  static void base_do_cancel_homing(int jno) {
      if (H[jno].homing) {
          H[jno].home_state = HOME_ABORT;
      }else if(H[jno].joint_in_sequence){
          H[jno].home_state = HOME_ABORT;
      }
  }
 ```
 then fixed.

andypugh, 2023-09-06:
Is this the same as https://github.com/LinuxCNC/linuxcnc/issues/2388 ?

yuyue2013, 2023-09-13:
yes， looks like same

andypugh, 2023-10-04:
Thanks for the fix. I will trust you that it works (it looks right)

closed 2023-10-04 by andypugh