LinuxCNC/linuxcnc #2630: Gmoccapy - Interupting homing

zz912, 2023-09-06:
RIP 2.9 Branche
Debian Bookworm
Oracle VM VirtualBox
Configuration: [Z0069_sim-001.zip](https://github.com/LinuxCNC/linuxcnc/files/12535504/Z0069_sim-001.zip)

If I move the GUI while homing, the homing of one axis is interrupted and another axis starts to be referenced.
[image]
Failed to simulate this in Axis.

hansu, 2023-12-30:
Does this also happen on a real machine with hardware limit switches?

Sigma1912, 2023-12-30:
Yes, this also happens on my real hardware.
Also happens when I switch to another desktop while homing.

zz912, 2023-12-30:
> Does this also happen on a real machine with hardware limit switches?

Yes