LinuxCNC/linuxcnc #2631: Gmoccapy - clicking in preview makes bug

zz912, 2023-09-06:
Clicking in preview makes bug:
[image]
```
[Gmoccapy][ERROR]  Found an error!
The following information may be useful in troubleshooting:

Traceback (most recent call last):
  File "/home/linaplast-zz/linuxcnc/linuxcnc-2.9-test/lib/python/rs274/glcanon.py", line 410, in inner
    return f(self, *args, **kw)
           ^^^^^^^^^^^^^^^^^^^^
  File "/home/linaplast-zz/linuxcnc/linuxcnc-2.9-test/lib/python/gremlin.py", line 496, in select_fire
    self.select(x, y)
  File "/home/linaplast-zz/linuxcnc/linuxcnc-2.9-test/lib/python/rs274/glcanon.py", line 595, in select
    buffer = glRenderMode(GL_RENDER)
             ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/OpenGL/GL/pointers.py", line 260, in glRenderMode
    raise error.GLError(
OpenGL.error.GLError: GLError(
	err = GL_STACK_OVERFLOW,
	description = b'stack overflow',
	result = 'glSelectBuffer too small to hold selection results'
)
 (gmoccapy:62)

```

c-morley, 2023-09-10:
This is lower in the code then Gmoccapy - affects Qtvcp and Gladevcp haven't heard about an AXIS report yet. I may have more details in the qtvcp section of forum...

c-morley, 2023-09-10:
I can't find it. IIRC they changed something in the underlying openGL python library (something we can't do of course) but it might give us a clue. I'll post if I find it.

gmoccapy, 2023-10-06:
I can reproduce this. affects all UIs that use gremlin.

gmoccapy, 2023-10-07:
Fixed 07.10.23 in 2.9

commit 5a455851 (2023-10-07): this fixes an error with buffer overflow when a user clicks very fast in the preview while a program is running. Fixes #2631
closed 2023-10-07 by gmoccapy