LinuxCNC/linuxcnc #2632: [2.9 AXIS/Lathe] Rigid Tapping (G33.1). "parse_error interp_error"

GTCLive, 2023-09-08:
Hi, 

I'm having problem with the Rigid Tapping (G33.1). I am not entirely sure how to debug linuxcnc as my tech skills are somehow limited and would appreciate if someone more savvy could confirm this is a bug. Included is a screenshot of the error and the gcode that is causing it.  

[image]

`O0001
N10 G18 G21 G7 G40 G49 G90 G80 G91.1 G64
N12 T16 G43 M6
N14 G54
N16 G94 G0 G90 X0. Z95.
N18 G97 S200. M3
N20 Z1.735
N22 G33.1 X0 Z-9.265 K.5
N24 G0 X0. Z95.
N26 X100.
N28 M5
N30 M2
N32 %`

**Environment:**
LINUXCNC - 2.9.0-pre1-1120-ga179f6e64
Linux debian 6.3.0-rt11-linuxcnc #2 SMP PREEMPT_RT Mon May  1 17:18:20 AEST 2023 x86_64 GNU/Linux

andypugh, 2023-09-08:
The problem might be the O0001 at the beginning.

GTCLive, 2023-09-08:
Hi, nah the problem is related to parsing the G33.1 (line 8). Remove that G33.1 out and it works fine, replace with G33 also works fine. Literally only complains when 33.1 (Rigid Tapping) is called. 

Removing the O0001 does not fix it unfortunately. Tried that before. No rush. When the devs feels like it. Somehow the operation complete regardless, just a nuisance really. 

Thanks guys 👍

GTCLive, 2023-09-10:
@andypugh Hi, I've set-up a quick environment and silenced Line 919 in gcodemodule.cc and installed **2.10.0~pre0**. 
A hint at what could possibly be going on is in the screenshot below, or error "_can only concatenate tuple (not "list") to tuple_".

[image]

Let me know if there is anything to look for, or if can do anything to help out ironing this bug.  Thanks.

itaib, 2023-09-12:
I believe the issue is a missed conversion from list to tuple in this commit: 70771d6de241dbedc06432b1f3b7335b7469231c

please check this branch/commit. I've only done a shallow test to see that the error message does not appear: 86c66c2ab30c2b58e81eea9eb7eca0644b31dd3b

commit 2a41bcbd (2023-09-15): resolves #2632 by using tuple instead of list in rigid_tap

the code cleanup in 70771d6de241dbedc06432b1f3b7335b7469231c did not include a conversion from list to tuple in the rigid_tap function, resulting in an error.
closed 2023-09-16 by c-morley