LinuxCNC/linuxcnc #2633: Python error after running Pncconf Wizard

HerrNamenlos123, 2023-09-08:
Hi there! Sorry if this is the wrong place to write this, but is indeed seems to me like a bug in the software.

I downloaded the latest available LinuxCNC version yesterday as the ISO file, and installed it in a VMWare Virtual Machine. Everything worked smoothly and I was able to run LinuxCNC (just for testing, not a real machine). I then ran the Pncconf Wizard, pretending to connect a Mesa 7i92 card. After clicking through all of the wizards, on the last page I just need to click `Done` and then this happens:

[image]

Sorry I cannot copy the error message out of the VM right now, but I think there is no need.

Even though I do not have a real Mesa card attached and I am running in a VM, this does indeed look like a bug and should never happen, no matter how wrong the numbers are you put into the wizard.

## Here are the steps I follow to reproduce the issue:

 1. Install http://www.linuxcnc.org/iso/linuxcnc-2.8.4-buster.iso in a VMWare VM
 2. Run the Pncconf Wizard, click through all pages, choose Mesa 7i92 with default Firmware
 3. Choose some step generators as required by the error messages
 4. Calculate default motor parameters and continue
 5. Press Done and the internal error appears

## This is what I expected to happen:

It works without errors, or there are proper errors that tell me what's wrong.

## This is what happened instead:

The error message

HerrNamenlos123, 2023-09-08:
Okay, it turns out that the error occurred because I chose Touchy UI and continued, and when it complained that all the axis step generators are missing, it also complained that an external hand wheel is required by Touchy. The thing is that even though it showed me this, it still let me continue, and some time later the error occurred. When going back, changing Touchy to AXIS and going through the rest again, it works.

Conclusion:
There must be a bug in some python implementation, because a series of UI interactions allowed me to get to a point where I am shown an internal python error message, that an average user has no idea what to do with. Said user would be stuck and likely give up.

I would expect a check that either blocks me from proceeding until I select the correct options, or shows a helpful error message in the end, that helps me identify the problem.

I am more or less a first-time user, so forgive me if I use the software incorrectly, but this is an issue I as a user got stuck with and would like to help improve the software by reporting it to get it fixed. Thank you for taking a look at this issue!

c-morley, 2023-09-10:
Thank you for the report.
Yes looks like this was fixed in 2.9 but not 2.8.
I pushed the fix to 2.8 but I doubt there will be another release of 2.8.
But you never know.

closed 2023-09-10 by c-morley