LinuxCNC/linuxcnc #2636: 2.10 Axis Sim

jethornton, 2023-09-14:
## Here are the steps I follow to reproduce the issue:

 1. Build and install master (2.10) on Debian 12
 2. Run the Axis sim
 3.

## This is what I expected to happen:
The Axis sim to start up

## This is what happened instead:
./cooling.hal:7: Pin 'iocontrol.0.lube' does not exist

## It worked properly before this:
LinuxCNC 2.9

## Information about my hardware and software:

 * I am using this Linux distribution and version Debian GNU/Linux 12 (bookworm)
 * I am using this kernel version Linux cave 6.1.0-9-rt-amd64 #1 SMP PREEMPT_RT Debian 6.1.27-1 (2023-05-08) x86_64 GNU/Linux

 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface  AXIS

After commenting out the cooling line I get 
Debug file information:
Note: Using POSIX realtime
link (updating variable file): No such file or directory
note: MAXV     max: 5.000 units/sec 300.000 units/min
note: LJOG     max: 5.000 units/sec 300.000 units/min
note: LJOG default: 0.250 units/sec 15.000 units/min
note: jog_order='XYZ'
note: jog_invert=set()
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/bin/axis", line 3972, in <module>
    get_coordinate_font(vars.dro_large_font.get())
  File "/usr/bin/axis", line 3850, in get_coordinate_font
    glnav.use_pango_font(coordinate_font, 0, 128)
  File "/usr/lib/python3/dist-packages/glnav.py", line 65, in use_pango_font
    glBitmap(1, 0, 0, 0, 0, h-d, bytearray([0]*4))
  File "/usr/lib/python3/dist-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/OpenGL/wrapper.py", line 700, in wrapperCall
    raise err
  File "/usr/lib/python3/dist-packages/OpenGL/wrapper.py", line 693, in wrapperCall
    result = wrappedOperation( *cArguments )
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
	err = 1285,
	description = b'out of memory',
	baseOperation = glBitmap,
	pyArgs = (
		1,
		0,
		0,
		0,
		0,
		16.0,
		<OpenGL.arrays._buffers.Py_buffer object at 0x7f8f30efbda0>,
	),
	cArgs = (
		1,
		0,
		0,
		0,
		0,
		16.0,
		<OpenGL.arrays._buffers.Py_buffer object at 0x7f8f30efbda0>,
	),
	cArguments = (
		1,
		0,
		0,
		0,
		0,
		16.0,
		<OpenGL.arrays._buffers.Py_buffer object at 0x7f8f30efbda0>,
	)
)
203908
203960
Stopping realtime threads
Unloading hal components
Note: Using POSIX realtime

itaib, 2023-09-14:
a cooling.lube reference was missed in 05943d180f7cc3efce617f8ede202c0b2e3bf3fa
commenting out this line fixes the issue for me https://github.com/LinuxCNC/linuxcnc/blob/29db881ed934174e26ac2f3390b01f74fcd20f80/configs/sim/axis/cooling.hal#L7

subsequently, this line can also be removed and the and2 instance count can be lowered to 1.  https://github.com/LinuxCNC/linuxcnc/blob/29db881ed934174e26ac2f3390b01f74fcd20f80/configs/sim/axis/cooling.hal#L4

rene-dev, 2023-09-14:
the opengl issue you are having is https://github.com/LinuxCNC/linuxcnc/issues/2510
I fixed the lube problem in https://github.com/LinuxCNC/linuxcnc/commit/6f82aeed0e8abe296685180c38769b56d37c6e00

closed 2023-09-14 by rene-dev