LinuxCNC/linuxcnc #2640: M4 issues / INI Documentation / pncconf

pcw-mesa, 2023-09-16:
It appears that M4 does not work with default pncconf created INI files
and possibly with other configuration utilities due to missing ini file settings

https://forum.linuxcnc.org/24-hal-components/50076-m4-doesn-t-work

Tested on 2.9 and 2.10

The INI documentation states that the SPINDLE_X  MAX_REVERSE_VELOCITY 
is set by default to SPINDLE_X MAX_VELOCITY, but this does not appear to 
be the case.

Unless MAX_REVERSE_VELOCITY is specifically set to a non-zero number
in the INI file, M4 will do nothing (0 commanded speed and spindle on = false)

In addition, the spindle INI pins do not appear on the ini HAL pin list

A simple fix would be for the configuration utilities to include the missing INI tokens

c-morley, 2023-09-17:
This seems a little deeper actually - Pncconf sets SPINDLE[0] entries for MAX_VELOCITY for say stepgens etc.
There has been no work in Pncconf that I know of to incorporate new motion related spindle INI settings  and it seems they clash.

A quick look in inispindle.cc seems the max_neg defaults to 0 and min_neg defaults to -1e99 - is that right?

Also min _neg and max_neg seem to be define by the opposite INI entry? 
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/ini/inispindle.cc#L80

Some comments of the idea of min/max of pos/neg might help - it's almost a double negative.
for instance is min_neg the lowest speed( minimum rotations ) or minimum absolute number? it's not clear enough IMHO.
slowest_neg/ fastest_neg would be better descriptors I think.

I think min_neg here should actually be negative?
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/ini/inispindle.cc#L67

I would say it is a coincidence that M3 worked.

jethornton, 2023-09-17:
This is not a Pncconf issue for sure because it affects any configuration that has a spindle section.
Creating a sim config with the stepconf tool and testing without a [SPINDLE_0] section and I get:

M3 S5000 give me spindle.0.speed-out 500 and spindle.0.forward ON as expected.

M4 give me spindle.0.speed-out -500 and spindle.0.reverse ON as expected.

After removing anything from the display section that says spindle then creating a spindle section with the following:

[SPINDLE_0]
MIN_VELOCITY = 0
MAX_VELOCITY = 500
INCREMENT = 500

M3 S5000 give me spindle.0.speed-out 500 and spindle.0.forward ON as expected.

M4 gives me spindle.0.speed-out 0 and spindle.0.reverse OFF which is not expected.

Adding the following to the spindle section:

MIN_REVERSE_VELOCITY = 100
MAX_REVERSE_VELOCITY = 250

And the spindle speed and direction pins respond to commands for forward and reverse as expected capping the reverse speed to -100 to -250.

Possibly this was introduced by Andy in this commit:
https://github.com/LinuxCNC/linuxcnc/commit/3d005f685d5abb7928bf37be66c946acc972636e

pcw-mesa, 2023-09-17:
So it looks like the default MAX_REVERSE_VELOCITY default setting is created from 
the the INI file DISPLAY section but not the SPINDLE section.

c-morley, 2023-09-18:
[DISPLAY] settings are usually GUI related which depending on the GUI could limit any other settings (at least while in manual mode) and disguise the problem.
Looking at the file I quoted, MAX_REVERSE_VELOCITY is searched in [SPINDLE_x] only

andypugh, 2023-09-23:
On Mon, 18 Sept 2023 at 16:11, pcw-mesa ***@***.***> wrote:

> Right, but it looks like only the DISPLAY setting creates the default
> MAX_REVERSE_VELOCITY value,
> not the SPINDLE_X settingMessage ID:
> ***@***.***>
>
This looks like a mistake made by me, I will try to fix it.

-- 
atp

c-morley, 2023-09-25:
pull request  #2659 is a fix for these problems or for discussion.

andypugh, 2023-09-28:
Do these changes to INI variables require a change in the update_ini script? 
In theory, probably not, as no _released_ version has used these (file not present in 2,8)

c-morley, 2023-09-28:
I don't think so. The clash is fixed so default limits are set. I doubt anyone purposely set them.

closed 2023-09-28 by andypugh