LinuxCNC/linuxcnc #2644: Unstable 'raster' test reports Test failed: name 'output' is not defined

petterreinholdtsen, 2023-09-19:
In https://github.com/LinuxCNC/linuxcnc/pull/2643, which added new PO files, one of the build tests failed with the following error:

```
Running test: /home/runner/work/linuxcnc/linuxcnc/tests/raster
*** /home/runner/work/linuxcnc/linuxcnc/tests/raster: XFAIL: test run exited with 1
************** result:
        error: Test failed: name 'output' is not defined
************** stderr:
        Note: Using POSIX non-realtime
        Note: Using POSIX non-realtime
**************
```

As far as I can tell, this test failure have nothing to do with the patch, and seem more to be due to a unstable testsuite check.

The test program set up a collection of HAL pins and use these to test the behaviour.  Could it be some race condition in the HAL processing?

rene-dev, 2023-10-12:
the raster test tries to link hal pins while the comp is loading.
should really call component_is_ready to wait for tje pins.
https://github.com/LinuxCNC/linuxcnc/blob/master/src/hal/halmodule.cc#L1829
except it cant, because it uses different hal python bindings.

rene-dev, 2023-10-12:
it cant be changed to use halmodule, as halmodule doesnt support hal ports.
https://www.youtube.com/watch?v=LFrdqQZ8FFc

grandixximo, 2026-10-05:
Fixed in 0b83d37656 (2026-08-24): the test now waits on the fast.threadbeat pin so the RT cycle has actually run before the outputs are checked, removing the CI race behind "name 'output' is not defined". Closing; reopen if it flakes again.

closed 2026-10-05 by grandixximo