LinuxCNC/linuxcnc #2651: Gmoccapy - preview - trajecory of remap is shown

zz912, 2023-09-21:
RIP 2.9 Branche
Debian Bookworm
Oracle VM VirtualBox
Configuration: [Remap_preview-002.zip](https://github.com/LinuxCNC/linuxcnc/files/12681864/Remap_preview-002.zip)

I made configuration with easy remap:
> o<ATC> sub
> 
> G53 G1 Z-195 f20000
> 
> G53 G1 Z0 f20000
> 
> o<ATC> endsub [10]
> m2

You can download it above.

The problem is that this ATC track is also displayed in the preview.
[image]

When I delete
N50T#<toolno>M6
the ATC track in the preview disappears.

It's even worse on the physical machine where only the ATC track is displayed. But I can't simulate that.

Source from forum:
https://www.forum.linuxcnc.org/40-subroutines-and-ngcgui/27872-gcode-preview-and-remap?start=10#277509

rene-dev, 2023-09-21:
you can check #<_task> to skip gcode in the preview, usually used for toolchanges.
https://linuxcnc.org/docs/html/gcode/overview.html#sub:system-parameters

zz912, 2023-09-21:
Could you help me more? What should I do? Try?

rene-dev, 2023-09-26:
The behaviour you are describing is documented here, along with the fix I proposed: https://linuxcnc.org/docs/html/remap/remap.html#remap:axis-preview-and-remapped-code-execution

closed 2023-09-26 by rene-dev