LinuxCNC/linuxcnc #2663: rcs_print_debug() has no effect in cms_cfg.cc:cms_create()

heeplr, 2023-09-28:
Some early debug statements like in [cms_create()](https://github.com/LinuxCNC/linuxcnc/blob/master/src/libnml/cms/cms_cfg.cc#L789) never get printed. 

It seems as if [the debug flags](https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/nml_intf/emcargs.cc#L45) are set after cms_create() runs (or maybe as if the flags get reset somewhere between emcGetArgs() and cms_create() and set again before deinit, which might be possible if unlikely).

Maybe I'll dive into that at some later point. For now leaving this issue for the record.

havardAasen, 2024-01-01:
@heeplr, did you fix this with #2717?

heeplr, 2024-01-01:
Yes, forgot to close this. Should be fixed now.

closed 2024-01-01 by heeplr