LinuxCNC/linuxcnc #2665: graphical limits frame not displayed on some systems

c-morley, 2023-09-28:
complaint:
https://forum.linuxcnc.org/38-general-linuxcnc-questions/50196-after-replacing-the-graphics-card-axis-qtdragon-and-gmoccapy-cannot-display-t#281800

Google indicates the opengl command GL_LINE_STIPPLE has problems.
Maybe a solid line frame would be better?

andypugh, 2023-09-29:
Skim-reading https://stackoverflow.com/questions/6017176/gllinestipple-deprecated-in-opengl-3-1 I wonder if GL_LINE_STRIP works? (The other answers there seem _massively_ complicated just to get a dotted line)

itaib, 2023-09-30:
2 options I can think of are
1) a solid textured line using blending and alpha mask to make it dotted
2) rendering multiple short lines / points.

I've tried approach 1 which seems to work, but there are some artifacts in the rendering. you can see a few gaps like the one circled in blue in this image.
the textured line is in green, the original sing GL_LINE_STIPPLE is in red (offset in X by 5 units)

[image]

I haven't yet tried the second approach, perhaps it will look better

andypugh, 2023-09-30:
The gaps would probably be less obvious in low-contrast red. 

On some levels this doesn't feel like it should be our job to fix, were it not for the fact that stipple has apparently been deprecated for a long time?

c-morley, 2023-10-01:
I think we should just consider solid lines

itaib, 2023-10-01:
On my rpi4, it seems the unsupported fallback is showing solid lines rather than not showing lines at all.

Perhaps the resolution is not doing anything without more info on the scope and just documenting it as a "known issue" with an easy self-fix (it's just a text file edit to make the lines solid, no recompile needed).
It looks like this issue affect only a few very old cards and could possibly be resolved by switching to a different driver.

Maybe revisit if it turns out to be more prevalent.

andypugh, 2023-10-01:
Is there any error return from the drivers if the line isn't drawn? 
ie, is there a way to know if a particular card/driver isn't showing the lines?

c-morley, 2023-10-01:
There is no error on my system.

I disagree on ignoring it.
It creates a bad user experience, as you can't easily tell when a loaded program is outside the limits.
Asking users to dig into core code that gets replaced on each update is not a good idea either.
The fix is easy and acceptable IMHO.
I could add the ability to change the color, so an integrator could pick an appropriate contrast, if that is a concern.

andypugh, 2023-10-01:
Lets just go with a solid but fairly dark red. Then I doubt that anyone will notice.

c-morley, 2023-10-02:
Do you wish to add this to 2.9?

vip74120, 2023-10-02:
It would be better if we can customize the color of the wireframe

c-morley, 2023-10-02:
Ya I have code - it was easy to add the ability to the underlying code.
Still up to the higher level GUI code to utilized it.
Defaults to red

vip74120, 2023-10-02:
> Is there any error return from the drivers if the line isn't drawn? ie, is there a way to know if a particular card/driver isn't showing the lines?

This phenomenon only occurs on older graphics cards (not necessarily). I happen to have one.

c-morley, 2023-10-02:
pushed fix to 2.9 +

closed 2023-10-02 by c-morley