LinuxCNC/linuxcnc #2667: Should machine translation to Russian be reverted?

petterreinholdtsen, 2023-09-30:
# The issue tracker is not a support forum

The LinuxCNC issue tracker is to report bugs in the software.
If you have a question about how to use the software, use one of the other methods detailed on our community support page: http://linuxcnc.org/community/

(delete this section before submitting your bug report)

## Here are the steps I follow to reproduce the issue:

 1.
 2.
 3.

## This is what I expected to happen:

## This is what happened instead:

## It worked properly before this:
(If the behavior changed after making a particular change in hardware or
software, describe the change you think is responsible.  E.g., "after upgrading
from LinuxCNC 2.7.3 to 2.7.4")

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
 * I am using this kernel version (shown by `uname -a`):
 * I am running ...
   * [ ] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`):
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc):
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):

petterreinholdtsen, 2023-10-01:
I decided to block xXx from contributing to LinuxCNC translations on
Weblate until I get a reply to my emails from him, to avoid making any revertion task any
larger.

I see IlyaKw has been updating at least some of the translations done by
xXx, no idea if it made it better or worse, but hope for the latter.  I
suspect this mean it is easier to update than to revert moving forward.
-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2023-10-01:
I believe the complete set last touched by xXx can be shown using https://hosted.weblate.org/search/linuxcnc/-/ru/?q=state%3A%3E%3Dtranslated+changed_by%3AxXx

andypugh, 2023-10-06:
Where do you think we are with this?

Actually, how does one access the Russian docs? 
http://linuxcnc.org/docs/stable/html/
Has links only for es, cn and fr.

petterreinholdtsen, 2023-10-06:
[Andy Pugh]
> Where do you think we are with this?

Last I heard on IRC, IlyaKw was going to have a look at the translations
done by xXx and correct the ones with errors.  Not sure how far he got
with it.

> Actually, how does one access the Russian docs? 
> http://linuxcnc.org/docs/stable/html/
> Has links only for es, cn and fr.

See <URL: http://linuxcnc.org/docs/devel/html/ >, the Weblate updates
are only in the master branch.

-- 
Happy hacking
Petter Reinholdtsen

IlyaKw, 2023-10-06:
I fixed the errors, this problem no longer exists.

andypugh, 2023-10-06:
> I fixed the errors, this problem no longer exists.

Thanks

petterreinholdtsen, 2023-10-06:
[IlyaKw]
> I fixed the errors, this problem no longer exists.

Very good.  The Weblate link with the translations xXx was the last to
touch still show 22 pages.  I assume this mean the translations on these
22 pages are all good.

-- 
Happy hacking
Petter Reinholdtsen

IlyaKw, 2023-10-06:
The contents of these lines are now correct

safeaegis, 2023-10-06:
russia is a terrorists state

IlyaKw, 2023-10-06:
[safeaegis]

Why are you writing this here?? For what ?

petterreinholdtsen, 2023-10-06:
[IlyaKw]
> The contents of these lines are now correct

Thank you very much for the help.  I guess it is time to close this
ticket. :)

-- 
Happy hacking
Petter Reinholdtsen

closed 2023-10-06 by andypugh

safeaegis, 2023-10-06:
russia is a terrorist country.
Any technical development on its territory is used to manufacture weapons and kill civilians in Ukraine. I will be grateful if the technical support of terrorists is stopped.
Even if it saves one innocent life, it will be worth it!
PS, I'm sorry that my last comment was not so detailed and you didn't fully understand it last time.
Sincerely, Mykhailo