LinuxCNC/linuxcnc #2670: PncConf - area "stepper scale" became black

zz912, 2023-10-02:
I've been getting this error since using version 2.9.

This is a completely random bug. When I turn PncConf off and on, the bug goes away.

Normal state:
[image]

Bug:
[image]

phillc54, 2023-10-02:
I don't see that issue on my machines. It may be that it caused by using the deprecated 'override_background'.
Attached is an experimental patch that uses CSS that you may like to try.

[pncconf_css.patch](https://github.com/LinuxCNC/linuxcnc/files/12780472/pncconf_css.patch)

zz912, 2023-10-02:
> I don't see that issue on my machines.

The bug is random, I couldn't find any dependency.

*.patch file is problem for me. I dont know, how to use it. Could you send me modified files? I will change it in my folders.

zz912, 2023-10-02:
I tried to make about 10 LCNC configurations and not once did this bug appear.

phillc54, 2023-10-02:
Is that after the changes?

zz912, 2023-10-02:
Yes.

phillc54, 2023-10-02:
Great, thank you. I will wait to see what Chris thinks of the changes.

6XoCtujg2C0gne, 2023-10-02:
> [Tracking image][https://github.com/notifications/beacon/ABFIKPM23JWRH2INRAX7H73X5KHKJA5CNFSM6AAAAAA5O4LO6KWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTH4AZWG.gif]

Folks,
As an FYI, this exact same scenario took place on my machine yesterday.
Regards
Andy

c-morley, 2023-10-02:
[image]
Is this what yours looks like?

Seems to work -just looks odd.

zz912, 2023-10-02:
I dont have problem with pink button. It worked me with normal red / white button.  I use default xfcee in Bookworm.

c-morley, 2023-10-02:
It's not the color. It's the fact only the '-' button was colored.

phillc54, 2023-10-02:
Is it appearing differently on different Linux distributions?
This is on Debian 12:
[image]

6XoCtujg2C0gne, 2023-10-03:
> [https://user-images.githubusercontent.com/2746571/272103767-ebc9536a-715c-46f6-8598-5ffd3aad3cef.png][Screenshot 2023-10-03 at 9 38 21 am][https://user-images.githubusercontent.com/2746571/272103767-ebc9536a-715c-46f6-8598-5ffd3aad3cef.png][https://user-images.githubusercontent.com/2746571/272103767-ebc9536a-715c-46f6-8598-5ffd3aad3cef.png]

I am running Debian 12 also pulled code and compiled.

There is still an issue with the A-Axis. It gets fouled up, cranky and has a fit! Ultimately just melts your screen....

phillc54, 2023-10-03:
> There is still an issue with the A-Axis. It gets fouled up, cranky and has a fit! Ultimately just melts your screen....

I am not really sure what you mean by this.
I see no issues on Debian 12 or Debian 11.

phillc54, 2023-10-03:
I didn't have css set up correctly if both encoders and steppers need changing to red.
This patch fixes that and also names the widgets in the glade files rather than in the python file.
[pncconf_css_1.patch](https://github.com/LinuxCNC/linuxcnc/files/12788119/pncconf_css_1.patch)

smoe, 2023-12-19:
@phillc54 , could you please prepare a pull request for your patch against 2.9? In principle I could happily do that for you but I likely get the authorships of that patch then wrong.

phillc54, 2023-12-20:
[done](https://github.com/LinuxCNC/linuxcnc/pull/2798)

zz912, 2023-12-30:
Can we close this Issue?

closed 2024-01-01 by zz912