LinuxCNC/linuxcnc #2692: linuxcncrsh: "get Tool_offset" returns uninitialized value

heeplr, 2023-10-09:
"get Tool_offset" seems to return an uninitialized [emcStatus->task.toolOffset.tran.z](https://github.com/LinuxCNC/linuxcnc/actions/runs/6459867877/job/17536468604?pr=2691#step:5:9051) as it returns [different values](https://github.com/LinuxCNC/linuxcnc/actions/runs/6459867877/job/17536468604?pr=2691#step:5:9051) for the same test, depending on the environment.
(Probably because there's no tooltable loaded in that test.)

heeplr, 2023-10-10:
58a1fd1 seems to fix one missing initialization but linuxcncrsh still returns seemingly uninitialized values with ```get tool_offset```. 

Compare [this test job](https://github.com/LinuxCNC/linuxcnc/actions/runs/6473936356/job/17577748645?pr=2691#step:5:9043) with gcc and [this test job](https://github.com/LinuxCNC/linuxcnc/actions/runs/6473936356/job/17577749192?pr=2691#step:5:9056) with clang from the same run.

rene-dev, 2023-10-11:
did you try what happens when you load a tooltable?

heeplr, 2023-10-11:
No, as it's not included in tests/linuxcncrsh (I'm only trying to get linuxcncrsh working rn).
I tried it with ```configure ... --enable-toolnml``` tho where it always get's initialized to 0.

Sigma1912, 2026-05-03:
Seems that this can be closed

commit 58a1fd15 (2023-10-10): initialize toolOffset, should close #2692