LinuxCNC/linuxcnc #2698: Compile error from today's PULL

6XoCtujg2C0gne, 2023-10-14:
## Here are the steps I follow to reproduce the issue:

 1. andy@file03:~/linuxcnc-current/src$ ./configure --disable-build-documentation-translation  --with-realtime=uspace --enable-non-distributable=yes 

 2. make
 3. This error occurs:

_Linking libpyplugin.so.0
c++ -std=gnu++17 -g -L/home/andy/linuxcnc-current/lib -Wl,-rpath,/home/andy/linuxcnc-current/lib -ltirpc  -lgpiod -Xlinker -export-dynamic -Wl,-soname,libpyplugin.so.0 -shared -o ../lib/libpyplugin.so.0 objects/emc/pythonplugin/python_plugin.o ../lib/liblinuxcncini.so -lstdc++ -lboost_python311 -L/usr/local/lib -lpython3.11 -ldl -lm
/usr/bin/ld: /usr/local/lib/libpython3.11.a(pythonrun.o): warning: relocation against `_Py_UnhandledKeyboardInterrupt' in read-only section `.text.unlikely'
/usr/bin/ld: /usr/local/lib/libpython3.11.a(abstract.o): relocation R_X86_64_PC32 against symbol `_Py_NotImplementedStruct' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make: *** [emc/pythonplugin/Submakefile:22: ../lib/libpyplugin.so.0] Error 1_

 

## This is what I expected to happen: Should have finished compilation.

## This is what happened instead:

_Linking libpyplugin.so.0
c++ -std=gnu++17 -g -L/home/andy/linuxcnc-current/lib -Wl,-rpath,/home/andy/linuxcnc-current/lib -ltirpc  -lgpiod -Xlinker -export-dynamic -Wl,-soname,libpyplugin.so.0 -shared -o ../lib/libpyplugin.so.0 objects/emc/pythonplugin/python_plugin.o ../lib/liblinuxcncini.so -lstdc++ -lboost_python311 -L/usr/local/lib -lpython3.11 -ldl -lm
/usr/bin/ld: /usr/local/lib/libpython3.11.a(pythonrun.o): warning: relocation against `_Py_UnhandledKeyboardInterrupt' in read-only section `.text.unlikely'
/usr/bin/ld: /usr/local/lib/libpython3.11.a(abstract.o): relocation R_X86_64_PC32 against symbol `_Py_NotImplementedStruct' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make: *** [emc/pythonplugin/Submakefile:22: ../lib/libpyplugin.so.0] Error 1_

## It worked properly before this:
It worked fine until I PULLed from GIT today

## Information about my hardware and software:

 * I am using this Linux distribution and version (often, shown by `lsb_release -a`):
andy@file03:~/linuxcnc-current/src$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

 * I am using this kernel version (shown by `uname -a`): Linux file03 6.1.0-11-rt-amd64 #1 SMP PREEMPT_RT Debian 6.1.38-4 (2023-08-08) x86_64 GNU/Linux

 * I am running ...
   * [X] A binary version from linuxcnc.org (including buildbot.linuxcnc.org)
   * [ ] A binary I built myself
   * [ ] A binary version from some other source besides linuxcnc.org
 * I am using this LinuxCNC version (shown in package manager or, for git versions, `scripts/get-version-from-git`): v2.10.0-pre0-2251-gc123203ab1
 * I am using this user interface (GUI) (e.g., AXIS, Touchy, gmoccapy, etc): AXIS
 * I am using this interface hardware vendor and chipset (e.g., parallel port, ethernet port, FPGA card):  BeeLink Mini S

6XoCtujg2C0gne, 2023-10-17:
Here's some additional detail:

_Compiling emc/pythonplugin/python_plugin.cc
emc/pythonplugin/python_plugin.cc: In constructor ‘PythonPlugin::PythonPlugin(_inittab*)’:
emc/pythonplugin/python_plugin.cc:298:22: warning: ‘void Py_SetProgramName(const wchar_t*)’ is deprecated [-Wdeprecated-declarations]
  298 |     Py_SetProgramName(program);
      |     ~~~~~~~~~~~~~~~~~^~~~~~~~~
In file included from /usr/local/include/python3.11/Python.h:94,
                 from /usr/include/boost/python/detail/wrap_python.hpp:178,
                 from /usr/include/boost/python/detail/prefix.hpp:13,
                 from /usr/include/boost/python/ssize_t.hpp:9,
                 from /usr/include/boost/python/object.hpp:8,
                 from emc/pythonplugin/python_plugin.hh:25,
                 from emc/pythonplugin/python_plugin.cc:19:
/usr/local/include/python3.11/pylifecycle.h:37:38: note: declared here
   37 | Py_DEPRECATED(3.11) PyAPI_FUNC(void) Py_SetProgramName(const wchar_t *);
      |                                      ^~~~~~~~~~~~~~~~~
emc/pythonplugin/python_plugin.cc: In member function ‘int PythonPlugin::configure(const char*, const char*)’:
emc/pythonplugin/python_plugin.cc:424:40: warning: ‘%s’ directive output may be truncated writing up to 4095 bytes into a region of size 4068 [-Wformat-truncation=]
  424 |         snprintf(pycmd, sizeof(pycmd), "import sys\nsys.path.append(\"%s\")", expandinistring);
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~
In file included from /usr/include/stdio.h:906,
                 from /usr/local/include/python3.11/Python.h:24:
In function ‘int snprintf(char*, size_t, const char*, ...)’,
    inlined from ‘int PythonPlugin::configure(const char*, const char*)’ at emc/pythonplugin/python_plugin.cc:424:10:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:54:35: note: ‘__builtin___snprintf_chk’ output between 31 and 4126 bytes into a destination of size 4096
   54 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   55 |                                    __glibc_objsize (__s), __fmt,
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   56 |                                    __va_arg_pack ());
      |                                    ~~~~~~~~~~~~~~~~~
emc/pythonplugin/python_plugin.cc: In member function ‘int PythonPlugin::configure(const char*, const char*)’:
emc/pythonplugin/python_plugin.cc:405:40: warning: ‘%s’ directive output may be truncated writing up to 4095 bytes into a region of size 4066 [-Wformat-truncation=]
  405 |         snprintf(pycmd, sizeof(pycmd), "import sys\nsys.path.insert(0,\"%s\")", expandinistring);
      |                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~~~~~
In function ‘int snprintf(char*, size_t, const char*, ...)’,
    inlined from ‘int PythonPlugin::configure(const char*, const char*)’ at emc/pythonplugin/python_plugin.cc:405:10:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:54:35: note: ‘__builtin___snprintf_chk’ output between 33 and 4128 bytes into a destination of size 4096
   54 |   return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
      |          ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   55 |                                    __glibc_objsize (__s), __fmt,
      |                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
   56 |                                    __va_arg_pack ());
      |                                    ~~~~~~~~~~~~~~~~~
Linking libpyplugin.so.0
c++ -std=gnu++17 -g -L/home/andy/linuxcnc-current/lib -Wl,-rpath,/home/andy/linuxcnc-current/lib -ltirpc  -lgpiod -Xlinker -export-dynamic -Wl,-soname,libpyplugin.so.0 -shared -o ../lib/libpyplugin.so.0 objects/emc/pythonplugin/python_plugin.o ../lib/liblinuxcncini.so -lstdc++ -lboost_python311 -L/usr/local/lib -lpython3.11 -ldl -lm
/usr/bin/ld: /usr/local/lib/libpython3.11.a(pythonrun.o): warning: relocation against `_Py_UnhandledKeyboardInterrupt' in read-only section `.text.unlikely'
/usr/bin/ld: /usr/local/lib/libpython3.11.a(abstract.o): relocation R_X86_64_PC32 against symbol `_Py_NotImplementedStruct' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make: *** [emc/pythonplugin/Submakefile:22: ../lib/libpyplugin.so.0] Error 1_

andypugh, 2023-10-18:
The only recent change in python_plugin.cc is https://github.com/LinuxCNC/linuxcnc/commit/5b0a6863927577c289cb06e464d80d6870d82b9c and there have been many successful builds since then. 

The fix suggested in the error message appears to already be there:
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/pythonplugin/Submakefile#L13

I wonder if what you are seeing is a consequence of https://github.com/LinuxCNC/linuxcnc/pull/2689/commits/c4855f163cdb89c8f479db86af9d09934a45e63b ? 

Maybe try going back to before then and see if you still get the failure?

rene-dev, 2023-10-19:
which compiler are you using?
did you try adding -fPIC here?
https://github.com/LinuxCNC/linuxcnc/blob/master/src/emc/pythonplugin/Submakefile#L22

6XoCtujg2C0gne, 2023-10-20:
I have been able to correct this by removing a upgraded version of Python. Noting that /usr/bin seems to be the default location for Python on Debian.