LinuxCNC/linuxcnc #2699: G71.X Path Issue

BSHoekstra, 2023-10-15:
**The cutter does not follow the G71.1 programed path correctly as described in the screen shots below.
Note:  This issue behaves differently in Linuxcnc versions 2.9 and 2.10.  Both versions are represented here.**  

## Here are the steps I follow to reproduce the issue:

G20 G18 G54 
G49 
G90 G92.1 G94 G64 p0.001
G8
G91.1 

; Contour
O200 SUB  	
G01 X0.75 Z0.05
G01 X0.75 Z-5.0
; G01 X1.2  **(In both V2.9, V2.10, In the 1st screen shot this line IS commented out and in the 2nd screen shot this line IS NOT commented out)**
O200 ENDSUB

; Tool Change Location
G0 X2
G0 Z-1.0
F6.0
M3 S1650
G40 T1 M6 G43 
G00 Z0.075

G71.1 Q200 X1.0 Z0.050 D0.040 I0.030 R0.125

M2

### **Using linuxcnc-uspace_2.9.1.3.g6684382de_amd64.deb**
[image]

### **Using linuxcnc-uspace_2.10.0-pre0-2069-g9de3c86d**
[image]

Thank you,

Barry

petterreinholdtsen, 2023-10-15:
[BSHoekstra]
> **The cutter does not follow the G71.1 programed path correctly as
> described in the screen shots below.  Note: This issue behaves
> differently in Linuxcnc versions 2.9 and 2.10.  Both versions are
> represented here.**

Perhaps you can make a test script demonstrating the problem, similar to
the ones in linuxcnc/tests/interp/g71-endless-loop,
linuxcnc/tests/interp/g71-loop, linuxcnc/tests/interp/g72-facing and
linuxcnc/tests/interp/g72-missing-iteration in the source code?

It make it a lot easier for the relevant developer to find and fix the
problem.
-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2023-10-15:
Are you 100% sure that both runs were using the same tool table?

BSHoekstra, 2023-10-15:
After working with this more, I did notice that the G71.1 cutter path was different in Linuxcnc versions 2.9 and 2.10.  This is probably not a major issue, but regardless, I thought I'd provide more information so you could see what I am seeing. 

Here is the simple test script to reproduce the issue.  Please notice the comments in the script showing changes necessary to reproduce the results. 

;Grill Handle
G20 G18 G54
G49
G90 G92.1 G94 G64 p0.001
G8
G91.1

; Contour
O200 SUB
G01 X0.75 Z0.05
G01 X0.75 Z-5.0
; G01 X1.2 **(In both V2.9, V2.10, In the 1st screen shot this line IS commented out and in the 2nd screen shot this line IS NOT commented out)**
O200 ENDSUB

; Tool Change Location
G0 X2
G0 Z-1.0
F6.0
M3 S1650
G40 T1 M6 G43
G00 Z0.075

G71.1 Q200 X1.0 Z0.050 D0.040 I0.030 R0.125

M2

Using linuxcnc-uspace_2.9.1-8-g2859fa8da_amd64.deb

[image]

   
Using linuxcnc-uspace_2.10.0-pre0-2069-g9de3c86d

[image]

Thank you,

Barry

P.S.  I checked both tool tables and they were the same.

BSHoekstra, 2023-10-16:
Note:  Food for thought....I do not think the V2.10 used here has the fix applied to resolve the "infinite do loop" defined in Issue #1146 based on a note in Issue #1146 that explained that the "infinite do loop" fix had not yet been uploaded to 2.10 from version 2.9 where this work was done.  So, I'm wondering if the the fix made to resolve the "infinite do loop" is creating this new issue since V2.10 appears to be working correctly as described earlier in the screen shots with the exception that the "infinite do loop" is an issue in this Version of 2.10.  I'll double verify this thought later tonight.

Thank you,

Barry

andypugh, 2023-10-16:
The fix commit is here:
https://github.com/LinuxCNC/linuxcnc/pull/2677/commits/03793cc9fa91f6d6f9f6cc2050922cc5a7149d95

And this is the relevant section of that file in Master:
https://github.com/petterreinholdtsen/linuxcnc/blob/master/src/emc/rs274ngc/interp_g7x.cc#L1194

So, the fix is in Master.