LinuxCNC/linuxcnc #2704: QT UIs pollute git

rene-dev, 2023-10-20:
The QT UIs seem to store some preferences in the sim folder.
after playing around with them, the git is dirty.
settings do not belong in a file tracked by version control.

```
modified:   configs/sim/qtplasmac/qtplasmac-metric.prefs
modified:   configs/sim/qtplasmac/qtvcp.prefs
```

andypugh, 2023-10-20:
Add to .gitignore?

rene-dev, 2023-10-20:
Gitignore would mean deleting the file. I don't know what the file does, or if it contains default values that are required.

phillc54, 2023-10-20:
Some of the preferences in these files show some options available in the GUI.

rene-dev, 2023-10-21:
if the files are deleted, the UI will recreate them. the config folder is the wrong place for a config file.

andypugh, 2023-10-21:
I think that the user's private config folder is the right place for a config file. 
It's just a feature of the RIP install that leads to them being created there. 

If the GUIs recreate them if missing, then I think that .gitignore is the answer.

c-morley, 2023-10-22:
git ignore is the answer - though I can't get it to work yet.
The preference file will end up in the user's config folder on a real machine.

c-morley, 2023-10-22:
@phillc54 can the pref files be uncommitted then ignored. ie Qtplasmac will work fine with no pref files the first time run?

phillc54, 2023-10-22:
Yes, there are reasonably sane defaults that it will self populate with.

c-morley, 2023-11-10:
@phillc54 you want to do this or I can - just so we can close the issue.

phillc54, 2023-11-11:
I will get to it either tomorrow or Monday.

phillc54, 2023-11-11:
Closed with [d7eff7f](https://github.com/LinuxCNC/linuxcnc/commit/d7eff7fabd3b80eedcfc17568f761dba5060f70c)

closed 2023-11-11 by phillc54