LinuxCNC/linuxcnc #2705: Glade not usable on Buster

rene-dev, 2023-10-21:
I was asked to open a separate issue on this.
Buster is holding a few things up, and we should drop support for it in master.

1. gcc8 doesnt support cpp20, which is really annoying
2. python 3.7 is really old, and long out of support https://devguide.python.org/versions/
3. https://github.com/LinuxCNC/linuxcnc/pull/2487
4. https://github.com/LinuxCNC/linuxcnc/pull/2671
5. busters glade uses python2, which doesnt allow glade panels to be edited: https://www.mail-archive.com/emc-developers@lists.sourceforge.net/msg21705.html
6. buster will be EOL  June 30th, 2024

for years we had to ship a special glade version for gtk2. finally wen can just use the glade version shipped with any modern OS.
I dont understand how we can claim that 2.9 or master "is supported on buster" when such a core component just doesn't work.

andypugh, 2023-10-22:
So, I cancel the 2.9 release?

rene-dev, 2023-10-22:
no, Im just saying that with 2.9 glade doesn't work on buster, due to the embedded interpreter.

andypugh, 2023-10-22:
It does work, with the right tricks. 

I took a 2.8 ISO install of Buster in a VM, updated to 2.9.1 using the LinuxCNC repo and then installed glade_3.22 from this deb:
http://www.linuxcnc.org/dists/buster/base/binary-amd64/glade_3.22.1-4_amd64.deb

[image]

piet-vr, 2023-10-23:
Hi,
Any ideas why Glade3.38.2 indicate all the Action widgets as deprecated?

c-morley, 2023-10-24:
https://lazka.github.io/pgi-docs/Gtk-3.0/classes/Action.html
https://gitlab.gnome.org/GNOME/gtk/-/issues/2590

explains a bit.

rene-dev, 2024-07-07:
closed by https://github.com/LinuxCNC/linuxcnc/pull/2671

closed 2024-07-07 by rene-dev