LinuxCNC/linuxcnc #2707: libtirpc found, but not usable

meerfrau, 2023-10-22:
## Here are the steps I follow to reproduce the issue:

```
./autogen.sh
./configure
make
```

rpc.h is at _/usr/include/tirpc/rpc/rpc.h_

## This is what I expected to happen:

libtirpc found
`pkg-config --cflags libtirpc` gives:
_-I/usr/include/tirpc_ 

## This is what happened instead:

`checking for usability of rpc/rpc.h... no`

applying [libtirpc.patch](https://aur.archlinux.org/cgit/aur.git/tree/libtirpc.patch?h=linuxcnc) doesn't change anything

## Information about my hardware and software:

* ArchLinux
* not built yet

andypugh, 2023-10-22:
What happens if you modify the paths here? 
https://github.com/LinuxCNC/linuxcnc/blob/master/src/configure.ac#L91

This is looking like a difference between Arch and Debian

grandixximo, 2026-10-05:
Closing as stale. The pkg-config fallback added in 3e809b6cd9 does work: current Debian (no /usr/include/rpc/rpc.h, libtirpc 1.3.x) builds through it daily, and the AUR package now just exports CPPFLAGS/CFLAGS/LDFLAGS instead of patching. If configure still fails on current Arch, please reopen with the config.log section.

closed 2026-10-05 by grandixximo