LinuxCNC/linuxcnc #2713: Tools canot be loaded after editting in (at least) qtdragon

c-morley, 2023-10-27:
In master branch:
Using a qtdragon sim, after startup, any tool can be loaded. if you edit the tool parameters in qtdragon's builtin tool editor, then you can not load any tool.

you get an error similar to:
emc/task/emctask.cc 68: interp_error: Requested tool 7 not found in the tool table
Requested tool 7 not found in the tool table

Qtdragon handles tool editing by writing directly to the tool file and asking linuxcnc to reload the tool table using python linuxcnc module's load_tool_table(). This is allowed only when linuxcnc is idle

I'm not sure how to debug any further.

c-morley, 2023-10-27:
Gmoccapy is effected too.

rene-dev, 2023-10-27:
axis is affected as well. click file->reload after that no tool can be loaded.

commit 844c66d3 (2023-10-27): load tooltable from ini, closes #2713
closed 2023-10-27 by rene-dev