LinuxCNC/linuxcnc #2715: test failure

rene-dev, 2023-10-29:
the CI has been failing for a while, anyone knows whats going on?
https://github.com/LinuxCNC/linuxcnc/actions/runs/6683612209/job/18159988194#step:7:9653

andypugh, 2023-10-29:
````
dh_installdocs --doc-main-package=linuxcnc-uspace --package=linuxcnc-doc-de
dh_installdocs: error: Cannot find (any matches for) "usr/share/doc/linuxcnc/LinuxCNC_Developer_de.pdf" (tried in ., debian/tmp)

	install -d debian/linuxcnc-doc-de/usr/share/doc/linuxcnc-doc-de
make[1]: *** [debian/rules:109: override_dh_installdocs-indep] Error 25
````
I am wondering why the .de docs get special treatment there? Especially as they appear to be the problem.

petterreinholdtsen, 2023-10-30:
[Andy Pugh]
> I am wondering why the .de docs get special treatment there?
> Especially as they appear to be the problem.

See <URL: https://github.com/LinuxCNC/linuxcnc/pull/2649 > for why.

I fail to understand why this fail on Bullseye but not when I test
locally.  Ideas how to reproduce is most welcome.

-- 
Happy hacking
Petter Reinholdtsen

havardAasen, 2023-10-30:
I looked through the log, and it seems that the German documentation isn't built, hence, its not found at install time. The logs has these messages

> 2023-10-29T13:20:15.0221576Z Get:401 http://deb.debian.org/debian bullseye/main amd64 po4a all 0.62-1 [1796 kB]
> [...]
> 2023-10-29T13:22:46.2294794Z checking for po4a... /usr/bin/po4a
> 2023-10-29T13:22:46.3426460Z configure: WARNING: po4a version too old, need version 0.67 or newer, not building translated docs

I don't know what caused it though, I can't find any changes in `.github/workflows/ci.yml`.

Actually, I don't understand how it could have worked to begin with, since we don't install a new enough version of po4a, In other jobs, we download a newer version with curl.

petterreinholdtsen, 2023-10-30:
[Håvard]
> Actually, I don't understand how it could have worked to begin with,
> since we don't install a new enough version of po4a, In other jobs, we
> download a newer version with curl.

Interesting idea.  I created a patch to ensure a new enough po4a is
available.  Perhaps it solve the issue?  We should know in an hour or
so.

-- 
Happy hacking
Petter Reinholdtsen

havardAasen, 2023-10-30:
I did some more interesting reading with the build logs, I found this line in the `package-indep` job, taken from the last successful build on the master branch <https://github.com/LinuxCNC/linuxcnc/actions/runs/6508327619>
> 2023-10-13T12:41:06.8475832Z Get:61 http://linuxcnc.org bullseye/base amd64 po4a all 0.67-2 [2198 kB]

Is the po4a package downloaded from different sources?

closed 2023-10-31 by rene-dev

petterreinholdtsen, 2023-10-31:
[Håvard]
> Is the po4a package downloaded from different sources?

Yes.  As a hack we download from snapshot.debian.org.  It is only needed
on older Debian versions, but at the moment no sensible version check is
done, so it is done on all Debian versions.

-- 
Happy hacking
Petter Reinholdtsen

andypugh, 2023-12-08:
http://linuxcnc.org/ bullseye/base amd64 po4a all 0.67-2 [2198 kB]

I don't think that file exists: https://www.linuxcnc.org/dists/bullseye/base/binary-amd64/

I will copy it across and see what happens.

havardAasen, 2023-12-10:
On 09.12.2023 00:50, andypugh wrote:
> http://linuxcnc.org/ <http://linuxcnc.org/> bullseye/base amd64 po4a all 0.67-2 [2198 kB]
> 
> I don't think that file exists: https://www.linuxcnc.org/dists/bullseye/base/binary-amd64/ <https://www.linuxcnc.org/dists/bullseye/base/binary-amd64/>
> 
> I will copy it across and see what happens.
> 

Yes, this might work. I can see that the file exist for Buster, which
also successfully builds the translations.

Though, something else to consider.
Do we really need to build translations for all of these distributions?
I can understand that we want to make sure that it is correct, but the
documentation? Wouldn't bookworm and sid be enough?

PR #2720 (2023-10-31): Make sure to install newer po4a when test building debs on github.
Fixes #2715.
commit 8a9a330e (2024-07-07): Make sure to install newer po4a when test building debs on github.

Fixes #2715.